dji-sdk/Onboard-SDK #768: Smaple compilation error

Hepan02, 2021-05-03:
I can't compile sample program on Debian 10. What I should do?
__________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster
__________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ gcc -v
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
__________________________________________________________________________________________________
pi@raspberrypi:~/Onboard-SDK/build $ cmake -version
cmake version 3.16.3
__________________________________________________________________________________________________
root@raspberrypi:/home/pi/Onboard-SDK/sample/platform/linux/telemetry/build# cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Onboard-SDK/sample/platform/linux/telemetry/build
__________________________________________________________________________________________________
When I use make i've got an errors:
root@raspberrypi:/home/pi/Onboard-SDK/sample/platform/linux/telemetry/build# make
Scanning dependencies of target djiosdk-telemetry-sample
[ 14%] Building CXX object CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_environment.cpp.o
[ 28%] Building CXX object CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o
[ 42%] Building C object CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/hal/osdkhal_linux.c.o
[ 57%] Building C object CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/osal/osdkosal_linux.c.o
[ 71%] Building CXX object CMakeFiles/djiosdk-telemetry-sample.dir/main.cpp.o
[ 85%] Building CXX object CMakeFiles/djiosdk-telemetry-sample.dir/telemetry_sample.cpp.o
[100%] Linking CXX executable djiosdk-telemetry-sample
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o: in function `LinuxSetup::LinuxSetup(int, char**, bool)':
(...)
make[2]: *** [CMakeFiles/djiosdk-telemetry-sample.dir/build.make:159: djiosdk-telemetry-sample] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/djiosdk-telemetry-sample.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
root@raspberrypi:/home/pi/Onboard-SDK/sample/platform/linux/telemetry/build# exit

dji-dev, 2021-05-04:
Agent comment from William Wong in <a href='https://djisdksupport.zendesk.com/agent/tickets/47239'>Zendesk ticket #47239</a>:<br><br>Hi Developer,

Thank you for reach out to DJI Developer Support.  Due to a Chinese Holiday there maybe a short delay in responding directly to your question - we apologize for any inconvenience this may cause you and look forward to addressing your ticket as soon as possible.

Warm Regards,
DJI Developer Support<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-05-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47239'>Zendesk ticket #47239</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

If you use the sample compilation environment directly, your compilation path seems incorrect, please perform the following steps under /home/pi/Onboard-SDK/
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/4GaQGm8wpXnOFVC9tyB1kfBKp' target='zd_attach'>inline-938112869.png</a><br><p><em><small>°°°</small></em></p>

Hepan02, 2021-05-05:
Dear Agent,
this is my  CMakeLists.txt. I think it's missed something link-directories.

```
cmake_minimum_required(VERSION 2.8)
project(djiosdk-telemetry-sample)

include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../../../core/inc)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/common)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/hal)
include_directories(include /home/pi/Onboard-SDK/sample/platform/linux/osal)
include_directories(include /usr/local/include)
include_directories(include /usr)
include_directories(include /home/pi/Onboard-SDK/)
include_directories(include /home/pi/Onboard-SDK/sample/core/inc)
include_directories(include /home/pi/Onboard-SDK/sample/core/src)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -g -O0")

FILE(GLOB SOURCE_FILES *.hpp *.cpp
        ${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_environment.cpp
        ${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_helpers.cpp
        ${CMAKE_CURRENT_SOURCE_DIR}/../hal/*.c
        ${CMAKE_CURRENT_SOURCE_DIR}/../osal/*.c
        )

if (OSDK_HOTPLUG)
    FILE(GLOB SOURCE_FILES ${SOURCE_FILES} ${CMAKE_CURRENT_SOURCE_DIR}/../hal/hotplug/*.c)
endif ()

add_executable(${PROJECT_NAME} ${SOURCE_FILES})
```

Examples in Onboard-SDK/build/bin - work properly. So the problem is with CMakeList. I don't have any target_link_libraries.

What I should add to : ~/Onboard-SDK/sample/platform/linux/telemetry/CMakeList.txt to compile that sample?

Best Regards.

closed 2021-05-07 by Hepan02