dji-sdk/Onboard-SDK #769: H20 Range finder API

ahmadhasan2k8, 2021-05-06:
I am trying to get access to the H20 camera's range finder data through OSDK. I think a similar Class is available in Mobile SDK here[](https://developer.dji.com/api-reference/android-api/Components/Camera/DJICamera_DJICameraLaserMeasureInformation.html)

Is there any API available to access H20's  Laser Range Finder data through OSDK?

dji-dev, 2021-05-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47345'>Zendesk ticket #47345</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, Currently OSDK does not support obtaining H20 camera's range finder data.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><p><em><small>°°°</small></em></p>

dji-kyle, 2021-06-25:
It is expected to support in November

closed 2021-06-25 by dji-kyle

ezobages, 2023-09-23:
hi @dji-dev  can you help me to add laser finder rng button in android msdk