dji-sdk/Onboard-SDK #770: Help creating CMakeList.txt for own project

Hepan02, 2021-05-07:
Dear Community.
I would like to compile my own project. I have error with `**make**`. `cmake .. `works properly.

I tried to compile the example in /Onboard-SDK/sample/platform/telemetry.
I did mkdir build and `cd build `in telemetry directory.

This is my CMakeList.txt
```
cmake_minimum_required(VERSION 2.8)
project(djiosdk-telemetry-sample)

# Compiler flags: link with pthread and enable C++11 support
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -g -O0")

# Tell the project where osdk-core is located
if(NOT ONBOARDSDK_SOURCE)
	set(ONBOARDSDK_SOURCE "/home/pi/Onboard-SDK/osdk-core")
	#set(ONBOARDSDK_SOURCE "/usr/local/include")

endif()

# Specify locations for osdk-core headers
include_directories(${ONBOARDSDK_SOURCE}/api/inc)
include_directories(${ONBOARDSDK_SOURCE}/utility/inc)
include_directories(${ONBOARDSDK_SOURCE}/hal/inc)
include_directories(${ONBOARDSDK_SOURCE}/protocol/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/linux/inc)
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../hal)
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../osal)
include_directories(include /usr/local/include)
include_directories(${ONBOARDSDK_SOURCE}/../sample/core/inc)

# User-code related project files
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../common)
FILE(GLOB SOURCE_FILES *.hpp *.cpp
	${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_environment.cpp
	${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_helpers.cpp
	main.cpp
	telemetry_sample.cpp
	)

if (OSDK_HOTPLUG)
    FILE(GLOB SOURCE_FILES ${SOURCE_FILES} ${CMAKE_CURRENT_SOURCE_DIR}/../hal/hotplug/*.c)
endif ()

# Target and linking
add_executable(${PROJECT_NAME} ${SOURCE_FILES})
target_link_libraries(${PROJECT_NAME} djiosdk-core)
```
This is the result after `make`
```
[ 20%] Linking CXX executable djiosdk-telemetry-sample
/usr/bin/ld: //usr/local/lib/libdjiosdk-core.a(dji_legacy_linker.cpp.o): in function `DJI::OSDK::LegacyLinker::legacyX5SEnableTask(void*)':
/home/pi/Onboard-SDK/osdk-core/api/src/dji_legacy_linker.cpp:69: **undefined reference to `DJI::OSDK**::Linker::getLocalSenderId()'
/usr/bin/ld: /home/pi/Onboard-SDK/osdk-core/api/src/dji_legacy_linker.cpp:71: **undefined reference** to `OsdkOsal_TaskSleepMs'
/usr/bin/ld: /home/pi/Onboard-SDK/osdk-core/api/src/dji_legacy_linker.cpp:72: **undefined reference** to `**DJI::OSDK**::Linker::isUSBPlugged()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/djiosdk-telemetry-sample.dir/build.make:129: djiosdk-telemetry-sample] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/djiosdk-telemetry-sample.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
I often have` [100%] Linking CXX executable djiosdk-telemetry-sample`.
The main problem is with "undefined reference".
I create CMakeList.txt according to: https://developer.dji.com/onboard-sdk/documentation/development-workflow/integrate-sdk.html

What I should do?

dji-dev, 2021-05-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47358'>Zendesk ticket #47358</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The lib: libdji-linker.a should be added(linked) to your project.
https://github.com/dji-sdk/Onboard-SDK/tree/master/osdk-core/linker
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/3UkmcwtipLn10VMVrytTfJyoq' target='zd_attach'>inline-703265209.png</a><br><p><em><small>°°°</small></em></p>

Hepan02, 2021-05-07:
Dear Agent,
I resolved most problems with "undefined reference". But still have some...
I added:
```
link_libraries(dji-linker)
link_libraries(${ONBOARDSDK_SOURCE}/linker/${ARCH}/lib/libdji-linker.a)

target_link_libraries(${PROJECT_NAME} PUBLIC djiosdk-core)
target_link_libraries(${PROJECT_NAME} PUBLIC  dji-linker)
target_link_libraries(${PROJECT_NAME} PUBLIC ${ONBOARDSDK_SOURCE}/linker/${ARCH}/lib/libdji-linker.a)
target_include_directories(${PROJECT_NAME} PUBLIC ${LIBUSB_1_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} PUBLIC ${LIBUSB_1_LIBRARIES})
target_link_libraries(${PROJECT_NAME} PUBLIC ${FFMPEG_LIBRARIES})
target_link_libraries(${PROJECT_NAME} PRIVATE advanced-sensing)
```
It helped but I have now these **errors**:
```
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x8): undefined reference to `OsdkLinux_UartInit'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0xc): undefined reference to `OsdkLinux_UartSendData'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x10): undefined reference to `OsdkLinux_UartReadData'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x14): undefined reference to `OsdkLinux_UartClose'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x18): undefined reference to `OsdkLinux_TaskCreate'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x1c): undefined reference to `OsdkLinux_TaskDestroy'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x20): undefined reference to `OsdkLinux_TaskSleepMs'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x24): undefined reference to `OsdkLinux_MutexCreate'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x28): undefined reference to `OsdkLinux_MutexDestroy'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x2c): undefined reference to `OsdkLinux_MutexLock'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x30): undefined reference to `OsdkLinux_MutexUnlock'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x34): undefined reference to `OsdkLinux_SemaphoreCreate'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x38): undefined reference to `OsdkLinux_SemaphoreDestroy'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x3c): undefined reference to `OsdkLinux_SemaphoreWait'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x40): undefined reference to `OsdkLinux_SemaphoreTimedWait'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x44): undefined reference to `OsdkLinux_SemaphorePost'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x48): undefined reference to `OsdkLinux_GetTimeMs'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x4c): undefined reference to `OsdkLinux_Malloc'
/usr/bin/ld: CMakeFiles/djiosdk-telemetry-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data+0x50): undefined reference to `OsdkLinux_Free'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/djiosdk-telemetry-sample.dir/build.make:144: djiosdk-telemetry-sample] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/djiosdk-telemetry-sample.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

Which library I should also add to the CMakeList.txt?

Hepan02, 2021-05-07:
@dji-dev Can you help me?

Hepan02, 2021-05-07:
@DJI-Jerry 
@dji-dev 

Can you help me?

dji-dev, 2021-05-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47358'>Zendesk ticket #47358</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

 [image]​
source code file: osdkhal_linux.c should be added to your compile environment:
https://github.com/dji-sdk/Onboard-SDK/tree/master/sample/platform/linux/hal
https://github.com/dji-sdk/Onboard-SDK/tree/master/sample/platform/linux/osal

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/q7cmrs9mvCar0EyP1tYFYXsRE' target='zd_attach'>inline1901882263.png</a><br><p><em><small>°°°</small></em></p>

Hepan02, 2021-05-08:
The proper CMakeList.txt for creating your own project (example based on telemetry sample):

```
cmake_minimum_required(VERSION 2.8)
project(djiosdk-telemetry-sample)

set(ARCH armv7)

# Compiler flags: link with pthread and enable C++11 support
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -g -O0")

# Tell the project where osdk-core is located
if(NOT ONBOARDSDK_SOURCE)
	set(ONBOARDSDK_SOURCE "/home/pi/Onboard-SDK/osdk-core")
endif()
set(CURRENT_CMAKE_MODULE_PATH ${ONBOARDSDK_SOURCE}/cmake-modules)

# Specify locations for osdk-core headers
include_directories(${ONBOARDSDK_SOURCE}/api/inc)
include_directories(${ONBOARDSDK_SOURCE}/utility/inc)
include_directories(${ONBOARDSDK_SOURCE}/hal/inc)
include_directories(${ONBOARDSDK_SOURCE}/protocol/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/linux/inc)
include_directories(${ONBOARDSDK_SOURCE}/linker/${ARCH}/inc)
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../hal)
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../hal/hotplug)
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../osal)
include_directories(include /usr/local/include)
include_directories(include /usr/local/lib)
include_directories(${ONBOARDSDK_SOURCE}/../sample/core/inc)

# User-code related project files
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/../common)
FILE(GLOB SOURCE_FILES *.hpp *.cpp
	${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_environment.cpp
	${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_helpers.cpp
	${CMAKE_CURRENT_SOURCE_DIR}/../hal/osdkhal_linux.c
	${CMAKE_CURRENT_SOURCE_DIR}/../osal/osdkosal_linux.c
	)

if (OSDK_HOTPLUG)
    FILE(GLOB SOURCE_FILES ${SOURCE_FILES} ${CMAKE_CURRENT_SOURCE_DIR}/../hal/hotplug/*.c)
endif ()

link_libraries(dji-linker)
link_directories(/usr/local/include /usr/local/lib)
link_libraries(${ONBOARDSDK_SOURCE}/linker/${ARCH}/lib/libdji-linker.a)

set(ADVANCED_SENSING_SOURCE_ROOT ${ONBOARDSDK_SOURCE}/advanced-sensing)
set(ADVANCED_SENSING_HEADERS_DIR
    ${ADVANCED_SENSING_SOURCE_ROOT}/api/inc
    ${ADVANCED_SENSING_SOURCE_ROOT}/platform/inc
    ${ADVANCED_SENSING_SOURCE_ROOT}/protocol/inc
    ${ADVANCED_SENSING_SOURCE_ROOT}/camera_stream/src
    ${ADVANCED_SENSING_SOURCE_ROOT}/ori-osdk-core/protocol/inc
    ${ADVANCED_SENSING_SOURCE_ROOT}/ori-osdk-core/hal/inc
    ${ADVANCED_SENSING_SOURCE_ROOT}/ori-osdk-core/platform/linux/inc
    )

set(CMAKE_MODULE_PATH ${CURRENT_CMAKE_MODULE_PATH})
find_package(LibUSB REQUIRED)
find_package(FFMPEG REQUIRED)
include_directories(${ADVANCED_SENSING_HEADERS_DIR})

# Target and linking
add_executable(${PROJECT_NAME} ${SOURCE_FILES})

target_link_libraries(${PROJECT_NAME} PUBLIC djiosdk-core)
target_link_libraries(${PROJECT_NAME} PUBLIC  dji-linker)
target_link_libraries(${PROJECT_NAME} PUBLIC ${ONBOARDSDK_SOURCE}/linker/${ARCH}/lib/libdji-linker.a)
target_link_libraries(${PROJECT_NAME} PUBLIC /usr/local/lib/libdjiosdk-core.a)
target_include_directories(${PROJECT_NAME} PUBLIC ${LIBUSB_1_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} PUBLIC ${LIBUSB_1_LIBRARIES})
target_link_libraries(${PROJECT_NAME} PUBLIC ${FFMPEG_LIBRARIES})
target_include_directories(${PROJECT_NAME} PUBLIC $<BUILD_INTERFACE:${ADVANCED_SENSING_HEADERS_DIR}>)
target_link_libraries(${PROJECT_NAME} PRIVATE advanced-sensing)
target_include_directories(${PROJECT_NAME} PUBLIC ${FFMPEG_INCLUDE_DIR})
target_link_libraries(${PROJECT_NAME} PUBLIC ${FFMPEG_LIBRARIES})
```

closed 2021-05-08 by Hepan02

highfreq, 2021-11-22:
Tested you CMakelists.txt for telemetry and it works, do you know what adjustments should be done to have it working for flight-control ??
I am trying to understand what each needs.

highfreq, 2021-12-09:
Please help me in linking what is missing when compilint only flight-control.
Cmake works but when i compile i get the error : undefined reference to `FlightSample::FlightSample(DJI::OSDK::Vehicle*)'

Could you please let me know what to add to CMakeLists.txt to have it working?

Please DO NOT link me to a useless github or forum answer, i already went troygh all of them and didn't find an answer.