dji-sdk/Onboard-SDK #771: "payloads_download_sample" sample application giving timeout on DJI Matric 300 RTK

akamilkhan, 2021-05-09:
We are trying to run "payloads_download_sample" sample application on DJI Matric 300 RTK. When we try to get filelist or filedata, we are getting a timeout.

```
[1372534.780]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[1372534.780]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[1372534.781]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[1372534.781]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE
```
The odd thing is, it starts working after we access the album from the handheld joystick. It keeps on working after that until we reboot the drone.

Any idea what might be going wrong?

Here are the complete logs from the application:
```
dji@manifold2:~/sdks/Onboard-SDK/build/bin$ sudo ./payloads_download_sample ./UserConfig.txt 
Read App ID
User Configuration read successfully. 

[1372503.316]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1372503.316]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = XXXXXXXXXXXXXX
[1372503.316]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.71
[1372503.316]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[1372503.317]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1372504.317]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[1372504.518]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[1372504.520]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[1372505.798]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[1372505.798]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[1372506.801]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[1372506.810]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[1372506.810]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[1372506.815]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1372506.815]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[1372506.817]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1372506.817]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[1372506.818]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[1372506.819]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[1372507.803]STATUS/1 @ firewallTask, L112: firewall task created ...
[1372507.815]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1372507.816]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = XXXXXXXXXXXXXX
[1372507.816]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.71
[1372507.840]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[1372507.840]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[1372507.842]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[1372507.842]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[1372507.842]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[1372507.842]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[1372508.642]STATUS/1 @ init, L254: Start advanced sensing initalization
[1372508.642]STATUS/1 @ activate, L1329: version 0x3040847

[1372508.653]STATUS/1 @ activate, L1367: Activation successful

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
a
[1372510.996]STATUS/1 @ main, L85: Play back mode setting......
[1372510.996]STATUS/1 @ main, L89: Get liveview right......
[1372511.000]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372511.000]STATUS/1 @ main, L93: Try to download file list  .......
[1372511.074]STATUS/1 @ fileListMonitorTask, L137: OSDK download monitor task created.
[1372513.080]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372513.081]STATUS/1 @ main, L146: Sample end ...

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
[1372517.076]STATUS/1 @ fileListMonitorTask, L153: curTimeMs:1372517076 refreshTimeMs:1372511074
[1372517.076]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[1372517.076]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[1372517.078]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[1372517.078]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE
a
[1372528.690]STATUS/1 @ main, L85: Play back mode setting......
[1372528.690]STATUS/1 @ main, L89: Get liveview right......
[1372528.695]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372528.695]STATUS/1 @ main, L93: Try to download file list  .......
[1372528.771]STATUS/1 @ fileListMonitorTask, L137: OSDK download monitor task created.
[1372530.778]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372530.778]STATUS/1 @ main, L146: Sample end ...

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
a[1372534.780]STATUS/1 @ fileListMonitorTask, L153: curTimeMs:1372534780 refreshTimeMs:1372528772
[1372534.780]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[1372534.780]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[1372534.781]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[1372534.781]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE

[1372534.810]STATUS/1 @ main, L85: Play back mode setting......
[1372534.810]STATUS/1 @ main, L89: Get liveview right......
[1372534.812]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372534.812]STATUS/1 @ main, L93: Try to download file list  .......
[1372534.870]STATUS/1 @ fileListMonitorTask, L137: OSDK download monitor task created.
[1372536.874]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[1372536.874]STATUS/1 @ main, L146: Sample end ...

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
b
[1372539.975]STATUS/1 @ main, L103: Download file number : 0
[1372539.975]STATUS/1 @ main, L106: Now try to download 0 media files from main camera.
[1372539.975]STATUS/1 @ main, L146: Sample end ...
[1372540.878]STATUS/1 @ fileListMonitorTask, L153: curTimeMs:1372540878 refreshTimeMs:1372534872
[1372540.878]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[1372540.879]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[1372540.879]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[1372540.879]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE
```

Thank in advance.

dji-dev, 2021-05-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47445'>Zendesk ticket #47445</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The odd thing is, it starts working after we access the album from the handheld joystick. It keeps on working after that until we reboot the drone.
Any idea what might be going wrong?
  -->OSDK and remote control (APP) cannot access the album at the same time.It is recommended that you verify this function of the OSDK without operating the APP. If the APP has access, please exit the album for a period of time (about 30s) and then access through the OSDK.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

lorenzave, 2021-05-10:
We have had the same experience. The SD card functionality seems to be still in beta state pretty unreliable. Not even putting the album mode takes it out of the state where it constantly timeouts.

dji-dev, 2021-05-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/47445'>Zendesk ticket #47445</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1. Make sure that the dual A usb connection is normal and the USB communication works.
2. If it still doesn't work, you can try to format the SD card first. and then verify this function after taking a new picture.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

akamilkhan, 2021-05-23:
Hi DJI,

Thank you for the response.

We tried again the things that you mentioned but nothing changed. We are still getting the timeout.

```
a[1372534.780]STATUS/1 @ fileListMonitorTask, L153: curTimeMs:1372534780 refreshTimeMs:1372528772
[1372534.780]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[1372534.780]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[1372534.781]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[1372534.781]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE
```

dji-kyle, 2021-07-07:
Sorry，Does this problem still exist? If it still exists and needs support from DJI , please send the specific exception information to dev@dji.com. 
Sorry, because of the mail system, we may not be able to effectively receive the reply message again.