dji-sdk/Onboard-SDK #776: InitPackageFromTopicList failed

AdamVen, 2021-05-28:
I'm attempting to modify the OSDK Telemetry sample to subscribe to different topics. However, after installing the modified version and selecting option "[C] Get telemetry data and save to file", I get the error `InitPackageFromTopicList failed`.

I modified the code to print the values being passed to InitPackageFromTopicList before the error handling. The extent of the modified code is:

```
bool
subscribeToDataAndSaveLogToFile(Vehicle* vehicle, int responseTimeout)
{
    signal(SIGINT, INThandler);
    // Telemetry: Verify the subscription
    ACK::ErrorCode subscribeStatus;
    subscribeStatus = vehicle->subscribe->verify(responseTimeout);
    if (ACK::getError(subscribeStatus) != ACK::SUCCESS)
    {
        ACK::getErrorCodeMessage(subscribeStatus, __func__);
        return false;
    }

    int       pkgIndex        = 0;
    int       freq            = 50;
    TopicName topicList50Hz[]  = {
             TOPIC_ALTITUDE_FUSIONED
            ,TOPIC_GPS_FUSED
            ,TOPIC_GPS_DATE
            ,TOPIC_GPS_TIME
    };

    int       numTopic        = sizeof(topicList50Hz) / sizeof(topicList50Hz[0]);
    bool      enableTimestamp = false;

    bool pkgStatus = vehicle->subscribe->initPackageFromTopicList(
            pkgIndex, numTopic, topicList50Hz, enableTimestamp, freq);
    std::cout <<"\n1. InitPackageFromTopicList" <<std::endl;
    // Print values to debug
    std::cout << pkgIndex <<std::endl;
    std::cout << numTopic <<std::endl;
    std::cout << topicList50Hz <<std::endl;
    std::cout << enableTimestamp <<std::endl;
    std::cout << freq <<std::endl;
    std::cout << pkgStatus <<std::endl;

    if (!(pkgStatus))
    {
      std::cout <<"1. InitPackageFromTopicList failed" <<std::endl;
      return pkgStatus;
    }

```
The output of which is:
[image]

The unmodified OSDK Telemetry sample runs successfully, so I added the same lines to debug, the output of which is:
[image]

Any ideas why pkgStatus returns 0 in the modified code but returns 1 in the telemetry sample code when the variables being passed to InitPackageFromTopicList seem to be fundamentally the same? Any help would be appreciated.

dji-dev, 2021-05-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48319'>Zendesk ticket #48319</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The TOPIC: TOPIC_GPS_DATE and TOPIC_GPS_TIME not support Freq: 50HZ, you can set the freq to 5.
 [image]​
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/EIVWK7AS1Snbz3kxfMJxgKPdF' target='zd_attach'>inline1523356683.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/lje7YflHlLOjQ941AQMCiGAo1' target='zd_attach'>inline1461923165.png</a>
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-05-31:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48319'>Zendesk ticket #48319</a>:<br><br>DJI developer website：
https://developer.dji.com/onboard-api-reference/group__telem.html
<br><p><em><small>°°°</small></em></p>