dji-sdk/Onboard-SDK #777: M300 cannot get compass status

github4529, 2021-05-28:
订阅 TOPIC_COMPASS拿到的是 Mag struct ,没法获得compass的状态

我看sample里面有M100的示例(getBroadcastData)可以获得compassStatus

dji-dev, 2021-05-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48325'>Zendesk ticket #48325</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

当前订阅中没有TOPIC获取compassStatus，可以结合使用的机型评估使用广播获取该数据。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-05-28:
不是说广播会占用大量带宽吗

dji-dev, 2021-05-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48325'>Zendesk ticket #48325</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

关于数据获取，除M100外均建议使用订阅获取数据，但是当前两种方式获取的数据不完全一致，订阅功能当前无法获取的compass状态，您可以评估订阅Mag数值来判断或者使用广播获取Compass状态。
/*!
 * @brief struct for data broadcast, return magnetometer reading
 *
 * @note returned value is calibrated mag data,
 * 1000 < |mag| < 2000 for normal operation
 */
typedef struct Mag
{
  int16_t x;
  int16_t y;
  int16_t z;
} Mag; // pack(1)

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-05-28:
请问这个三元组如何用来判断compass的状态

dji-kyle, 2021-06-24:
可以大致根据值 1000 < |mag| < 2000来判断，该状态值无法精准提供compass状态。

github4529, 2021-06-24:
> 可以大致根据值 1000 < |mag| < 2000来判断，该状态值无法精准提供compass状态。
请问是否是
|mag| = sqrt(x*x +y*y +z*z)