dji-sdk/Onboard-SDK #780: advancedSensing->startMainCameraStream

Napalys, 2021-06-02:
Dear DJI team,

It seems like the startMainCameraStream function has an x amount of time delay between when the callback happens and real life, thus the whole video stream is being delayed.

So the question arises, do you try to compensate for such delay with telemetry or end up getting a frame from the past (delayed by X amount of time) while the GPS / Gimbal telemetry would be in real-time?

Best regards,
Napalys

dji-dev, 2021-06-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48529'>Zendesk ticket #48529</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the drone type, camera type and OSDK version?
And about the video delay, please refer to the developer website:
https://developer.dji.com/document/8801892e-22be-4c16-a621-87b5fbe18211
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/HJdcN5vgq9hHMJ2rhcOJQdAvi' target='zd_attach'>inline509772811.png</a>
<br><p><em><small>°°°</small></em></p>

Napalys, 2021-06-03:
The drone is M300, camera H20T, OSDK 4.1.

So as far as I understand you do not try to synchronize the stream with telemetry data such as - GPS data / Gimbal Data?
If the answer is not, is it possible to synchronize telemetry data with the video stream?

dji-dev, 2021-06-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48529'>Zendesk ticket #48529</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Video stream from OSDK API is not synchronized with telemetry data. You can evaluate and subscribe the telemetry data to synchronize with stream by yourself.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2021-06-07 by Napalys