dji-sdk/Onboard-SDK #781: 激光测距

JunGenius, 2021-06-04:
你好，请问下 OSDK,能否获取激光测距信息 ？ 非常感谢

dji-dev, 2021-06-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48593'>Zendesk ticket #48593</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

抱歉，当前OSDK还不支持获取激光测距信息。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

JunGenius, 2021-06-04:
> Agent comment from kyle.cai in [Zendesk ticket #48593](https://djisdksupport.zendesk.com/agent/tickets/48593):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 抱歉，当前OSDK还不支持获取激光测距信息。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

请问下有没有什么别的好的办法能获取到测距信息 ？我非常需要这个信息， 非常感谢。

dji-kyle, 2021-06-24:
OSDK预计11月份发布支持H20激光测距版本，当前MSDK可以获取。

closed 2021-06-24 by dji-kyle