dji-sdk/Onboard-SDK #782: SDK for DJI mini 2

AlexanderPretel, 2021-06-05:
Hi, has the SDK for the mini 2 been contemplated?

dji-dev, 2021-06-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48639'>Zendesk ticket #48639</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

OSDK does not support DJI mini 2, Are you inquiry about Mobile SDK ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

AlexanderPretel, 2021-06-07:
> Agent comment from kyle.cai in Zendesk ticket #48639

Hello again.  Yes, specifically with the mobile SDK for Android and IOS, do
you have any idea when they release that SDK from the mini 2 and mavic 2S?

-----------------

Cordialmente,

Alexander Pretel Diaz

Estudiante Ingeniería Topográfica
Representante Estudiantil ante el Comité del Programa Académico de
Ingeniería Topográfica
Facultad de Ingeniería
Universidad del Valle
Tel: 3166215704

Santiago de Cali-Colombia
2019-2
V. M.

dji-kyle, 2021-06-24:
Please visit Mobile SDK Repository：
https://github.com/dji-sdk/Mobile-SDK-iOS
https://github.com/dji-sdk/Mobile-SDK-iOS

closed 2021-06-24 by dji-kyle