dji-sdk/Onboard-SDK #785: OSDK+遥控器操控无人机

github4529, 2021-06-09:
请问有没有办法通过OSDK+遥控器遥控无人机

看到了订阅主题里有一个TOPIC：TOPIC_RC_WITH_FLAG_DATA好像是获取遥控器操作的？

有没有sample演示如何操控无人机的

dji-dev, 2021-06-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48799'>Zendesk ticket #48799</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

OSDK与遥控器控制无人机有控制权的切换问题，不能同时使用OSDK和遥控器来控制，OSDK可以通过订阅消息TOPIC：TOPIC_RC_WITH_FLAG_DATA获取遥控器的部分信息，可以结合使用机型来对应获取相关遥测数据来实现OSDK的控制。遥测数据获取可以参考sample：
https://github.com/dji-sdk/Onboard-SDK/tree/master/sample/platform/linux/telemetry

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-06-09:
> Agent comment from kyle.cai in [Zendesk ticket #48799](https://djisdksupport.zendesk.com/agent/tickets/48799):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> OSDK与遥控器控制无人机有控制权的切换问题，不能同时使用OSDK和遥控器来控制，OSDK可以通过订阅消息TOPIC：TOPIC_RC_WITH_FLAG_DATA获取遥控器的部分信息，可以结合使用机型来对应获取相关遥测数据来实现OSDK的控制。遥测数据获取可以参考sample：
> https://github.com/dji-sdk/Onboard-SDK/tree/master/sample/platform/linux/telemetry
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

可以通过OSDK的订阅获取遥控器的RC数据，然后可以怎么模拟来控制无人机的动作呢
telemetry示例展示了如何获取rc数据

github4529, 2021-06-09:
遥控器-->OSDK telemetry RC-->无人机

dji-kyle, 2021-06-24:
> 可以通过OSDK的订阅获取遥控器的RC数据，然后可以怎么模拟来控制无人机的动作呢
> telemetry示例展示了如何获取rc数据

OSDK可以使用对应的飞行控制API来控制无人机动作，可参考flight-control 相关示例。

closed 2021-06-24 by dji-kyle