dji-sdk/Onboard-SDK #788: 能否判断当前相机模式

github4529, 2021-06-10:
看了代码 相机的工作模式分为拍照 录像等

能不能知道当前相机为可见光还是红外，还有红外分屏

github4529, 2021-06-10:
以及能否通过OSDK切换可见光/红外模式

M300、H20/Z30等摄像头

dji-dev, 2021-06-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48853'>Zendesk ticket #48853</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

OSDK不支持获取当前使用的镜头类型，可以通过API设置镜头（H20）：
  /*! @brief
  *  Change the camera stream source from one payload device. (Beta API)
  *
  *  @platforms M300
  *  @note Only support for payload device : H20/H20T
  *  @param pos point out which camera to output the H264 stream
  *  @param source change to be the target camera of the payload, ref to
  *        LiveView::LiveViewCameraSource
  *  @return Errorcode of liveivew, ref to DJI::OSDK::LiveView::LiveViewErrCode
  */
  LiveViewErrCode changeH264Source(LiveViewCameraPosition pos, LiveViewCameraSource source);

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

closed 2021-06-24 by dji-kyle