dji-sdk/Onboard-SDK #791: DJI Assistant2 的模拟器

github4529, 2021-06-15:
OSDK中的飞控sample可以通过模拟器模拟飞行，能通过遥控器模拟飞行吗

如果无人机与遥控器距离过远，是否无法接收到遥控器的控制指令，此时能否通过订阅TOPIC关于RC的指令接收到遥控器的操作，再通过OSDK进行相关的控制

dji-dev, 2021-06-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/48969'>Zendesk ticket #48969</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

1、DJI Assistant2中打开模拟器后，可以使用OSDK或遥控器控制飞机进行仿真模拟

2、遥控器与无人机过远导致无线信号太弱存在数据丢失时，可能收不到遥控器的指令。OSDK获取遥测指令是通过飞机获取的，飞机上记录遥控器的指令也需要确保遥控器信号能传递到飞机上。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-06-15:
> Agent comment from kyle.cai in [Zendesk ticket #48969](https://djisdksupport.zendesk.com/agent/tickets/48969):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 1、DJI Assistant2中打开模拟器后，可以使用OSDK或遥控器控制飞机进行仿真模拟
> 
> 2、遥控器与无人机过远导致无线信号太弱存在数据丢失时，可能收不到遥控器的指令。OSDK获取遥测指令是通过飞机获取的，飞机上记录遥控器的指令也需要确保遥控器信号能传递到飞机上。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

运行sample里的flight-control，auto-landing的时候，到0.7米的高度就不动了，要开遥控器才能降落。无法脱离遥控器直接进行飞控吗

dji-kyle, 2021-06-24:
需要使用强制降落或确认降落API，参考demo中的C选项。

closed 2021-06-24 by dji-kyle