dji-sdk/Onboard-SDK #792: ./MatriceSeries_ConfigTool --usb-port /dev/ttyACM0 --c./MatriceSeries_ConfigTool: error while loading shared libraries: libavcodec.so.57: cannot open shared object file: No such file or directory

Avistadrone, 2021-06-18:
Hello,

I have been trying to activate the flag that let the drone fly with the USB A- USB A cable connected but I am getting the following error:

./MatriceSeries_ConfigTool --usb-port /dev/ttyACM0 --config-file /home/anuc/.ros/UserConfig.txt --usb-connected-flight on
./MatriceSeries_ConfigTool: error while loading shared libraries: libavcodec.so.57: cannot open shared object file: No such file or directory

Everything in the SDK work nice but the MatriceSreis_ConfigTool, it seems like the only package that depends on libavcodec.57.

I am running this in an intel Nuc I7 with Ubuntu 16.04.
```
dpkg -l "*avcodec*" | grep ii
ii  libavcodec-dev:amd64            7:2.8.17-0ubuntu0.1 amd64        FFmpeg library with de/encoders for audio/video codecs - development files
ii  libavcodec-ffmpeg-extra56:amd64 7:2.8.17-0ubuntu0.1 amd64        FFmpeg library with additional de/encoders for audio/video codecs
```

dji-dev, 2021-06-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49140'>Zendesk ticket #49140</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

It is recommanded to check the compile file and link the lib path, refer to OSDK Cmake file:
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/wRPnf87rCT9qBUUTDzq57z6LN' target='zd_attach'>inline-2107270008.png</a>
<br><p><em><small>°°°</small></em></p>

Avistadrone, 2021-06-18:
Hello,

I know, but the problem it's that the version installed in Ubuntu 16.04, the version in which you have runned the test for the Onboard SDK ROS, is libavcodec56. So it's quite surprising that the configTool depends on verision 57.

Ffmpeg is found, but 56 version.
-- Found FFmpeg  installed in the system.
--  - Includes: 
--  - Libraries: //usr/lib/x86_64-linux-gnu/libavcodec.so;//usr/lib/x86_64-linux-gnu/libavformat.so;//usr/lib/x86_64-linux-gnu/libavutil.so;//usr/lib/x86_64-linux-gnu/libswscale.so;//usr/lib/x86_64-linux-gnu/libvorbis.so;//usr/lib/x86_64-linux-gnu/libvorbisenc.so;//usr/lib/x86_64-linux-gnu/libtheora.so;//usr/lib/x86_64-linux-gnu/libz.so

Avistadrone, 2021-06-21:
@dji-dev  any news?

I followed this link: https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html

And when ffmpeg is installed as you said in the tutorial I get that the library is the newest version available for Ubuntu 16.04, which is the version in which you have done all the tutorial...

1. Install Development Tools

    C ++ compiler: GCC 5.4.0/5.5.0 version
    CMake: 2.8 and above
    Linux: Ubuntu 16.04 (If you need to use advanced sensor functions, please use the Libusb library)

```
sudo apt-get install libavcodec-dev libswresample-dev
[sudo] password for anuc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libavcodec-dev is already the newest version (7:2.8.17-0ubuntu0.1).
libswresample-dev is already the newest version (7:2.8.17-0ubuntu0.1).
The following packages were automatically installed and are no longer required:
  libgavl1 libhdf5-mpi-dev libhdf5-openmpi-10 libhdf5-openmpi-dev libignition-cmake-dev libignition-math4
  libignition-math4-dev libignition-msgs libignition-msgs-dev libignition-transport4 libignition-transport4-dev
  libqwt-headers libqwt-qt5-6 libqwt-qt5-dev libyaml-dev libzip-dev libzip4 libzmq3-dev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 112 not upgraded.
```

closed 2021-06-23 by Avistadrone

eduardofamendes, 2022-09-26:
Hello,

Any updates on that @dji-dev? I am facing the same problem. 

@Avistadrone, did you manage to resolve it?

Thanks,

Eduardo Mendes