dji-sdk/Onboard-SDK #793: 飞控相关操作的取消

github4529, 2021-06-18:
两个相关的cpp，flight_control应该相对新一点，其实差不多

都是同步/阻塞的，如果下发返航指令，途中想要取消返航，是否同步的无法实现，需要使用异步的指令才能取消

dji-dev, 2021-06-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49144'>Zendesk ticket #49144</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

不好意思，不太明白您的意思。OSDK中分别提供了同步和异步接口来发送取消返航指令。
 [image]

您是指​startGohome后，接口阻塞无法发送cancel指令吗？可以结合应用考虑开线程处理。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/DfI1wo3m7xnvhkfZeGI2SDupy' target='zd_attach'>inline2029296511.png</a>
<br><p><em><small>°°°</small></em></p>

dji-kyle, 2021-06-24:
如果问题依旧存在，请再确认重启，谢谢

closed 2021-06-24 by dji-kyle

github4529, 2021-06-24:
降落与返航的Sync-API，Async和Sync的cancel API都可以取消。
降落与返航的Async-API，使用Async和Sync的cancel API都不能取消