dji-sdk/Onboard-SDK #796: 使用M300，如何设置在一个航点的悬停时间？

GreyYu6458, 2021-06-28:
你好，我正在使用M300，并使用航线功能模块。

我现在需要在某个航点位置停留一个自定义的时间，但我没有发现任何一个Action符合我的要求。

在Onboard-SDK是否支持这个功能，具体涉及哪些API？

dji-dev, 2021-06-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49493'>Zendesk ticket #49493</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

M300的waypoint v2功能没有直接action可以设置航点停留动作，可以通过
DJIWaypointV2ActionActuatorAircraftControlOperationTypeFlyingControl 参数：isStartFlying来设置停止或飞行动作，waittime可以设置恢复时间。
单位为 0.1s， 数据类型为uint8，设置停留时间最大25.5s。若需悬停更长时间可以考虑使用pause暂停。

具体动作的使用可以参考相关帖：
https://djisdksupport.zendesk.com/hc/zh-cn/articles/4402367493785-OSDK-waypoint-v2-sample%E4%BB%A3%E7%A0%81%E8%B7%9F%E8%AF%BB

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

GreyYu6458, 2021-06-28:
那我是否可以通过多设置几个DJIWaypointV2ActionActuatorTypeAircraftControl以超过25.5s的限制呢？
我之前通过DJIWaypointV2ActionTriggerTypeSampleReachPoint和pause、resume函数，发现航线的轨迹会变得比较奇怪。

dji-kyle, 2021-07-07:
理论上是可以支持的，关于pause、resume后航线轨迹变得奇怪可能需要具体排查一下。
抱歉，因为邮件系统的原因，再次回复的问题，可能无法直接收到导致问题有遗漏。后续关于SDK的问题，建议您可以直接邮件发送至dev@dji.com进行确认。

GreyYu6458, 2021-07-09:
好的 谢谢你的回复

closed 2021-07-09 by GreyYu6458