dji-sdk/Onboard-SDK #798: Commanding M210v2 to fly at lower height

archielee, 2021-06-29:
On our previous M210 RTK, we were able to use advanced flight control (`Control::flightCtrl(CtrlData data)`) in SDK v3.8.1 to control the drone. We used it to do tasks at low heights.

Moving to the M210 RTK v2 and SDK v4.0.1, we have found that the drone will not move below 70 cm. Is this a safety feature? If so, is there a way for us to bypass this check in the SDK?

dji-dev, 2021-06-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49539'>Zendesk ticket #49539</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

When the downward vision obstacle avoidance function is turned on, there will be 0.7m protection, you can try to turn off the downward vision function, and use the VERTICAL_THRUST mode in the vertical control.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

archielee, 2021-06-29:
When doing this, do we have to use VERTICAL_THRUST? Is VERTICAL_POSITION no longer available?

archielee, 2021-07-07:
Confirmed that I can disable it from the DJI Pilot app and still perform vertical position control. Closing this issue, thanks!

closed 2021-07-07 by archielee