dji-sdk/Onboard-SDK #800: 设置相机的ShutterSpeed失败

github4529, 2021-07-01:
M300，H20T
跑sample里的payloads，sync和async的a示例，都无法设置shutterspeed，获取也不行
```
[1635295.739]STATUS/1 @ setExposureModeSyncSample, L122: Get exposure mode = 1
[1635295.739]STATUS/1 @ setExposureModeSyncSample, L129: Set exposure mode = 2
[1635295.757]ERRORLOG/1 @ setExposureModeSyncSample, L134: Set exposure mode error. Error code : 0xD00000000E3
[1635295.758]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Camera
[1635295.758]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Camera received invalid parameters
[1635295.758]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Check the validity of the parameter
[1635295.778]STATUS/1 @ setShutterSpeedSyncSample, L245: Set shutterSpeed = 33
[1635295.799]ERRORLOG/1 @ setShutterSpeedSyncSample, L250: Set shutterSpeed parameter error. Error code : 0xD00000000E3
[1635295.799]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Camera
[1635295.799]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Camera received invalid parameters
[1635295.799]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Check the validity of the parameter
[1635295.799]ERRORLOG/1 @ setShutterSpeedSyncSample, L252: The shutter speed can be set only when the camera exposure mode is Shutter mode or Manual mode. The shutter speed should not be set slower than the video frame rate when the camera's mode is RECORD_VIDEO.
[1635297.799]STATUS/1 @ main, L223: Sample end ...

```

dji-dev, 2021-07-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/49737'>Zendesk ticket #49737</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

可以先将代码中的ExposureMode设置为CameraModule::ExposureMode::EXPOSURE_MANUAL，再设置shutterspeed。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-07-05:
> Agent comment from kyle.cai in [Zendesk ticket #49737](https://djisdksupport.zendesk.com/agent/tickets/49737):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 可以先将代码中的ExposureMode设置为CameraModule::ExposureMode::EXPOSURE_MANUAL，再设置shutterspeed。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

你好，现在运行示例(payloads/main_sync)，相机模式为1，PROGRAM_AUTO，无法设置为其他模式(2,3,4) SHUTTER_PRIORITY/APERTURE_PRIORITY/EXPOSURE_MANUAL，
Error code：0xD00000000E3，请问可能是什么原因

github4529, 2021-07-05:
> Agent comment from kyle.cai in [Zendesk ticket #49737](https://djisdksupport.zendesk.com/agent/tickets/49737):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 可以先将代码中的ExposureMode设置为CameraModule::ExposureMode::EXPOSURE_MANUAL，再设置shutterspeed。
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> 
> 
> _°°°_

我又运行了下async的sample，能设置1，4 ，设置为2，3时失败。

github4529, 2021-07-05:
遥控器设置H20T为广角 可设置为Program_auto
设置为变焦，可设置PROGRAM_AUTO, EXPOSURE_MANUAL
其他情况均失败。

github4529, 2021-07-06:
#801

closed 2021-07-06 by github4529