dji-sdk/Onboard-SDK #805: payloads的对焦功能

github4529, 2021-07-08:
M300 +H20T
```
| [f] Set camera tap zoom point                                  |
|     Vice camera : Z30, H20/H20T(zoom mode) etc.                |
| [g] Set camera zoom parameter                                  |
|     Vice camera : Z30, H20/H20T(zoom mode) etc.                |

f
[2234254.878]STATUS/1 @ setTapZoomPointSyncSample, L575: Set tap zoom enable  = 1
[2234256.241]ERRORLOG/1 @ setTapZoomPointSyncSample, L578: Set tap zoom enable fail. Error code : 0x3
[2234256.242]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234256.242]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234256.242]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234256.242]ERRORLOG/1 @ setTapZoomPointSyncSample, L580: It is only supported Z30 camera or H20/H20T zoom mode.
[2234261.242]STATUS/1 @ setTapZoomPointSyncSample, L575: Set tap zoom enable  = 1
[2234262.606]ERRORLOG/1 @ setTapZoomPointSyncSample, L578: Set tap zoom enable fail. Error code : 0x3
[2234262.606]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234262.606]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234262.606]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234262.606]ERRORLOG/1 @ setTapZoomPointSyncSample, L580: It is only supported Z30 camera or H20/H20T zoom mode.
[2234267.606]STATUS/1 @ main, L227: Sample end ...

g
[2234311.855]STATUS/1 @ setZoomSyncSample, L646: Attention : It is only supported by X5, X5R and X5S camera on Osmo withlens Olympus M.Zuiko ED 14-42mm f/3.5-5.6 EZ, Z3 camera, Z30 camera.
[2234311.855]STATUS/1 @ getOpticalZoomFactorSync, L822: Get lens data from cmd request.
[2234313.214]ERRORLOG/1 @ setZoomSyncSample, L653: Get zoom parameter fail. Error code : 0x3
[2234313.214]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234313.214]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234313.214]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234313.215]STATUS/1 @ setZoomSyncSample, L655: Attention : It is only supported by X5, X5R and X5S camera on Osmo withlens Olympus M.Zuiko ED 14-42mm f/3.5-5.6 EZ, Z3 camera, Z30 camera.
[2234317.215]STATUS/1 @ setZoomSyncSample, L646: Attention : It is only supported by X5, X5R and X5S camera on Osmo withlens Olympus M.Zuiko ED 14-42mm f/3.5-5.6 EZ, Z3 camera, Z30 camera.
[2234317.215]STATUS/1 @ getOpticalZoomFactorSync, L822: Get lens data from cmd request.
[2234318.583]ERRORLOG/1 @ setZoomSyncSample, L653: Get zoom parameter fail. Error code : 0x3
[2234318.584]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234318.584]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234318.584]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234318.584]STATUS/1 @ setZoomSyncSample, L655: Attention : It is only supported by X5, X5R and X5S camera on Osmo withlens Olympus M.Zuiko ED 14-42mm f/3.5-5.6 EZ, Z3 camera, Z30 camera.
[2234322.584]STATUS/1 @ startZoomSyncSample, L622: Attention : It is only supported by X5, X5R and X5S camera on Osmo withlens Olympus M.Zuiko ED 14-42mm f/3.5-5.6 EZ, Z3 camera, Z30 camera.
[2234322.584]STATUS/1 @ startZoomSyncSample, L626: Start continuous optical zoom parameters : direction=0, speed=78
[2234323.953]ERRORLOG/1 @ startZoomSyncSample, L631: Start continuous zoom fail. Error code : 0x3
[2234323.953]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234323.953]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234323.953]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234331.953]STATUS/1 @ stopZoomSyncSample, L686: Stop continuous optical zoom.
[2234333.321]ERRORLOG/1 @ stopZoomSyncSample, L689: Stop continuous zoom fail. Error code : 0x3
[2234333.321]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : System
[2234333.321]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Request time out
[2234333.321]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Try again or check the status of the target object.
[2234333.322]STATUS/1 @ main, L227: Sample end ...
```

dji-dev, 2021-07-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50046'>Zendesk ticket #50046</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

相机管理功能建议检查：
1、双A USB连接且通信正常。
2、API参数Index与物理挂载口一致，飞机朝向左侧为主云台，对应PAYLOAD_INDEX_0
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/NoTGzxlkxD10FvSHllNktxdIe' target='zd_attach'>inline1039843216.png</a>
<br><p><em><small>°°°</small></em></p>

github4529, 2021-07-09:
> Agent comment from kyle.cai in [Zendesk ticket #50046](https://djisdksupport.zendesk.com/agent/tickets/50046):
> 
> 尊敬的开发者，
> 
> 您好！感谢您联系DJI大疆创新。
> 
> 相机管理功能建议检查：
> 1、双A USB连接且通信正常。
> 2、API参数Index与物理挂载口一致，飞机朝向左侧为主云台，对应PAYLOAD_INDEX_0
> [image]​
> 
> 感谢您的理解与支持,祝您生活愉快!
> 
> Best Regards，
> DJI大疆创新SDK技术支持
> [inline1039843216.png](https://djisdksupport.zendesk.com/attachments/token/NoTGzxlkxD10FvSHllNktxdIe)
> 
> 
> _°°°_

您好，能正常运行e示例，无法运行fg示例
建议检查项都无问题，

github4529, 2021-07-09:
> Agent comment from kyle.cai in [Zendesk ticket #50046](https://djisdksupport.zendesk.com/agent/tickets/50046):
>
> 尊敬的开发者，
>
> 您好！感谢您联系DJI大疆创新。
>
> 相机管理功能建议检查：
> 1、双A USB连接且通信正常。 [...]

挂载了单云台，使用index为PAYLOAD_INDEX_0运行其他多项相关的API都正常

github4529, 2021-07-12:
sample里对焦功能默认设置未PAYLOAD_INDEX_1，修改为PAYLOAD_INDEX_0后正常

closed 2021-07-12 by github4529