dji-sdk/Onboard-SDK #806: cmake .. always return -- Configuring incomplete, errors occurred!

zijian98, 2021-07-08:
Hi, I have downloaded the latest Onboard-SDK-4.1 and I am trying to Configure ROS Development Environment. However, when I tried these commands:

`$ mkdir build
 $ cd build
 $ cmake ..
 $ sudo make -j7 install`

it always shows -- Configuring incomplete, errors occurred! whenever I tried cmake .. 

Linux environment was setup correctly and I have installed the development tool.
Linux version: Ubuntu 20.04.2.0 LTS
cmake version: 3.16.3

How do I fix this?

Thanks

dji-dev, 2021-07-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50051'>Zendesk ticket #50051</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the terminal logs about the compiling errors ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

zijian98, 2021-07-12:
> Agent comment from kyle.cai in [Zendesk ticket #50051](https://djisdksupport.zendesk.com/agent/tickets/50051):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> Could you please provide the terminal logs about the compiling errors ?
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team
> 
> 
> _°°°_

Problem is solved. Thanks

closed 2021-07-12 by zijian98