dji-sdk/Onboard-SDK #807: 获取由航点任务拍摄得到的照片

GreyYu6458, 2021-07-09:
你好 我现在正在使用M300并挂载了H20T，现在我通过OSDK设计航线，并成功拍摄了照片，但照片默认保存在了 H20T的存储卡中。
请问有没有办法通过OSDK得到拍摄的照片？

dji-dev, 2021-07-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50099'>Zendesk ticket #50099</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

OSDK可以下载H20 SD卡中的照片。
参考sample：
https://github.com/dji-sdk/Onboard-SDK/blob/master/sample/platform/linux/payloads/download_sample.cpp

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

GreyYu6458, 2021-07-09:
好的 谢谢你的帮助

GreyYu6458, 2021-07-19:
@dji-dev 

你好 我尝试运行了download_sample.cpp这个sample
该例程在尝试下载文件列表时报错了，而且我已经挂载了H20T 
程序执行状况如下:

Read App ID
User Configuration read successfully. 

[3168040.214]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3168040.214]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7800ASJ15
[3168040.214]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.74
[3168040.214]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3168040.215]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3168041.215]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3168041.415]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[3168041.415]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3168042.679]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[3168042.679]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[3168043.680]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[3168043.691]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[3168043.691]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[3168043.693]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3168043.693]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[3168043.695]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3168043.695]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[3168043.697]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[3168043.697]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[3168044.681]STATUS/1 @ firewallTask, L112: firewall task created ...
[3168044.695]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3168044.695]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH7800ASJ15
[3168044.695]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.74
[3168044.706]ERRORLOG/1 @ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[3168044.706]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[3168044.706]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[3168044.707]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!
[3168044.707]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3168044.707]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3168045.507]STATUS/1 @ init, L254: Start advanced sensing initalization
[3168045.507]STATUS/1 @ activate, L1329: version 0x304084A

[3168045.514]STATUS/1 @ activate, L1367: Activation successful

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
a
[3168048.164]STATUS/1 @ main, L85: Play back mode setting......
[3168048.164]STATUS/1 @ main, L89: Get liveview right......
[3168048.166]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[3168048.166]STATUS/1 @ main, L93: Try to download file list  .......
[3168048.234]STATUS/1 @ fileListMonitorTask, L137: OSDK download monitor task created.
[3168050.242]STATUS/1 @ printErrorCodeMsg, L936: Execute successfully.
[3168050.242]STATUS/1 @ main, L146: Sample end ...

| [a] Download main camera filelist                              |
| [b] Download main camera filedata from case a                  |
| [q] Quit                                                       |
[3168054.234]STATUS/1 @ fileListMonitorTask, L153: curTimeMs:3168054234 refreshTimeMs:3168048234
[3168054.234]ERRORLOG/1 @ fileListMonitorTask, L154: downloadMonitorTask timeout!! device type : 1 index: 0
[3168054.234]STATUS/1 @ SendAbortPack, L957: SendAbortPack
[3168054.234]STATUS/1 @ fileListReqCB, L34: ##[Download main camera file list] : ret = 255 
[3168054.234]STATUS/1 @ fileListMonitorTask, L164: Finish req filelist task cause of timeout, reset downloadState to be DOWNLOAD_IDLE`
`
导致这样的原因可能是什么？我其他功能都运行正常，包括航线和拍照。

GreyYu6458, 2021-07-20:
我已经解决了该问题 感谢你的回复 @dji-dev

closed 2021-07-20 by GreyYu6458

wuxiangchao, 2025-01-02:
> 我已经解决了该问题 感谢你的回复 @dji-dev

你好，请问你是怎么解决的这个问题？我也遇到了同样的问题