dji-sdk/Onboard-SDK #809: DJI OSDK RTK定位信息获取精度问题

hhuxuchang, 2021-07-11:
利用DJI::OSDK::DataBroadcast
Telemetry::RTK getRTKInfo ()
读取无人机的定位信息，为什么经度纬度高度都只是到小数点后四位，精度没有理论上那么高，请问这个问题应该怎么解决呢？

hhuxuchang, 2021-07-11:
[image]

hhuxuchang, 2021-07-11:
存下来就是这样子，精度很低，飞行了200多米，RTK由于位数少，就显得无人机没有动

dji-dev, 2021-07-12:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50145'>Zendesk ticket #50145</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

请问您使用的机型是什么？

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

hhuxuchang, 2021-07-12:
我们的机型是M300 RTK 用的网络RTK ---- 回复的原邮件 ***@***.***>发送日期2021年07月12日 14:54 ***@***.***> ***@***.******@***.***>主题Re: [dji-sdk/Onboard-SDK] DJI OSDK RTK定位信息获取精度问题 (#809)

Agent comment from kyle.cai in Zendesk ticket #50145:尊敬的开发者，

您好！感谢您联系DJI大疆创新。

请问您使用的机型是什么？

感谢您的理解与支持,祝您生活愉快!

Best Regards，

DJI大疆创新SDK技术支持

°°°

[

{

"target": "https://github.com/dji-sdk/Onboard-SDK/issues/809#issuecomment-878020149",

"url": "https://github.com/dji-sdk/Onboard-SDK/issues/809#issuecomment-878020149",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

hhuxuchang, 2021-07-13:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    我们使用的是DJI M300 RTK
    
        
            
        
    
    
     
    
        
                    ***@***.***
        
    
    
        签名由网易邮箱大师定制