dji-sdk/Onboard-SDK #810: H20T相机是否不支持获取拍照模式

github4529, 2021-07-12:
可以设置拍照模式为SINGLE INTERVAL和SR等模式，看代码H20T设置模式时需要特殊处理。
获取拍照模式没有特殊处理，获取失败

osdk-core/modules/src/payload/dji_camera_module.cpp中
setShootPhotoModeSync的API能正常使用，而getShootPhotoModeSync不能正常使用，返回失败

dji-dev, 2021-07-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50196'>Zendesk ticket #50196</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

感谢您的反馈，此问题我们已经复现，当前显示Command not supported，我们已经提交进一步确认和修复。若有任何进展，我们也将与您同步。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-07-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50196'>Zendesk ticket #50196</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

此问题我们收到相关工程师的更新回复，此问题为bug，我们将在后续的master分支上更新修复，您可以继续关注后续master分支版本的修复提交说明。
再次抱歉给您相关开发工作带来的不便！

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

closed 2022-03-31 by dji-kyle