dji-sdk/Onboard-SDK #811: waypointV2运行失败

github4529, 2021-07-14:
M300+H20T
在DJI Assistant 2下使用模拟器模拟飞行
跑waypointV2示例，提示没有rtk，
在subscribe下添加了rtk相关TOPIC也不行。是否需要实际飞行才能测试
```
./djiosdk-waypoint-v2-sample
Read App ID
User Configuration read successfully.

[2729073.899]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2729073.899]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B3R0E
[2729073.899]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.74
[2729073.899]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2729073.899]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2729074.899]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[2729075.099]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[2729075.099]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2729075.116]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[2729075.116]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[2729076.118]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[2729076.133]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[2729076.133]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[2729076.135]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2729076.135]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[2729076.137]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[2729076.137]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[2729076.139]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[2729076.139]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[2729077.119]STATUS/1 @ firewallTask, L112: firewall task created ...
[2729077.144]STATUS/1 @ getDroneVersion, L1702: ret = 0
[2729077.144]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH8500B3R0E
[2729077.144]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.74
[2729077.163]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[2729077.163]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[2729077.165]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[2729077.165]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[2729077.165]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[2729077.165]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[2729077.966]STATUS/1 @ init, L254: Start advanced sensing initalization
[2729077.966]STATUS/1 @ activate, L1329: version 0x304084A

[2729077.984]STATUS/1 @ activate, L1367: Activation successful

[2729079.005]STATUS/1 @ RegisterMissionEventCallback, L433: register result of geting mission event pushing : 1

[2729079.006]STATUS/1 @ RegisterMissionStateCallback, L452: register result of geting mission state pushing : 1

[2729079.025]STATUS/1 @ verify, L244: Verify subscription successful.
[2729079.042]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729079.042]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729079.042]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729079.042]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729079.042]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0

[2729079.045]STATUS/1 @ startPackage, L345: Start package 0 result: 0.
[2729079.045]STATUS/1 @ startPackage, L347: Package 0 info: freq=1, nTopics=6.

[2729079.050]STATUS/1 @ runWaypointV2Mission, L187: Set up subscription successfully!
[2729080.043]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729080.043]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729080.044]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729080.044]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729080.044]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0
[2729080.058]STATUS/1 @ initMissionSetting, L299: Init mission setting successfully!
[2729081.045]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729081.045]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729081.045]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729081.045]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729081.045]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0
[2729081.058]STATUS/1 @ missionEncode, L135: mis_upload_start_index:0, mis_upload_end_index:7, upload_len:132
[2729081.354]STATUS/1 @ uploadWaypointMission, L316: Upload waypoint v2 mission successfully!
[2729082.047]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729082.048]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729082.048]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729082.048]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729082.048]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0
[2729082.360]STATUS/1 @ downloadWaypointMission, L332: Download waypoint v2 mission successfully!
[2729083.050]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729083.050]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729083.050]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729083.051]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729083.051]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0
[2729083.363]STATUS/1 @ getActionRemainMemory, L739: Total memory is:65535

[2729083.363]STATUS/1 @ getActionRemainMemory, L740: Remain memory is:65535

[2729083.363]STATUS/1 @ getActionRemainMemory, L506: get waypoint v2 action remain memory successfully!
[2729083.363]STATUS/1 @ ActionsEncode, L374: upload_action_ID:0
[2729083.363]STATUS/1 @ ActionsEncode, L374: upload_action_ID:1
[2729083.363]STATUS/1 @ ActionsEncode, L374: upload_action_ID:2
[2729083.363]STATUS/1 @ ActionsEncode, L374: upload_action_ID:3
[2729083.363]STATUS/1 @ ActionsEncode, L374: upload_action_ID:4
[2729083.363]STATUS/1 @ ActionsEncode, L376: total_len:60
[2729083.366]STATUS/1 @ uploadWaypointActions, L348: Upload waypoint v2 actions successfully!
[2729083.370]STATUS/1 @ getActionRemainMemory, L739: Total memory is:65535

[2729083.371]STATUS/1 @ getActionRemainMemory, L740: Remain memory is:65390

[2729083.371]STATUS/1 @ getActionRemainMemory, L506: get waypoint v2 action remain memory successfully!
[2729084.050]STATUS/1 @ updateMissionState, L58: missionStatePushAck->commonDataVersion:10
[2729084.051]STATUS/1 @ updateMissionState, L59: missionStatePushAck->commonDataLen:6
[2729084.051]STATUS/1 @ updateMissionState, L60: missionStatePushAck->data.state:0x0
[2729084.051]STATUS/1 @ updateMissionState, L61: missionStatePushAck->data.curWaypointIndex:0
[2729084.051]STATUS/1 @ updateMissionState, L62: missionStatePushAck->data.velocity:0
[2729084.380]ERRORLOG/1 @ startWaypointMission, L357: Start waypoint v2 mission ErrorCode:0x110000020005
[2729084.380]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : WaypointV2
[2729084.380]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : use rtk_data     but rtk is not connected or rtk_data is invalid
[2729084.380]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : none
[2729084.393]STATUS/1 @ verify, L244: Verify subscription successful.
[2729084.414]STATUS/1 @ reset, L551: Reset Subscription Successful.

[2729084.834]STATUS/1 @ ~AdvancedSensing, L186: Destroy task for M300's liveview!
[2729084.834]STATUS/1 @ ~AdvancedSensing, L194: Destroy task for M300's advanced sensing!

```

dji-dev, 2021-07-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50271'>Zendesk ticket #50271</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

模拟器无法模拟RTK，模拟器飞行时，可以在遥控器（APP）上关闭RTK。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-07-14:
遥控器关闭RTK可以在模拟器上模拟飞行

closed 2021-07-14 by github4529