dji-sdk/Onboard-SDK #812: 树莓派如何编译OSDK

github4529, 2021-07-19:
树莓派4
OSDK 最新分支
M300
```
mkdir build
cd build
cmake ..
sudo make
```
```
pi@raspbian:~$ uname -a
Linux raspbian 4.19.79-OPENFANS+20191013-v8 #1 SMP PREEMPT Sun Oct 13 16:06:07 CST 2019 aarch64 GNU/Linux

pi@raspbian:~/Onboard-SDK/build$ sudo make
[ 37%] Built target advanced-sensing
[ 91%] Built target djiosdk-core
make[2]: *** No rule to make target '../osdk-core/linker/armv8/lib/libdji-linker.a', needed by 'bin/djiosdk-flightcontrol-sample'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:222: sample/platform/linux/flight-control/CMakeFiles/djiosdk-flightcontrol-sample.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
pi@raspbian:~/Onboard-SDK/build$

dji-dev, 2021-07-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/50519'>Zendesk ticket #50519</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

可以检查一下linker库是否有放到编译环境，是否匹配平台环境。
 [image]​

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持<br><a href='https://djisdksupport.zendesk.com/attachments/token/sG6xTkuW8yTGZMs4AVtKIwB4E' target='zd_attach'>inline727691816.png</a>
<br><p><em><small>°°°</small></em></p>

github4529, 2021-07-20:
github下载相关linker库，已解决

closed 2021-07-20 by github4529