dji-sdk/Onboard-SDK #816: stm32f4 discovery uart2 output wrong code

askariz, 2021-07-31:
[image]

as you can see , whatever I change baudrate to 9600/115200/230400 is the same result .

askariz, 2021-07-31:
[image]

askariz, 2021-07-31:
[image]

also uncheck RTS and DTR

dji-dev, 2021-08-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51089'>Zendesk ticket #51089</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

界面显示像是显示格式不对，建议看看是不是有设置显示格式，将分包显示勾选去掉试试看。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

closed 2022-03-31 by dji-kyle