dji-sdk/Onboard-SDK #817: Zoom is not applied after calling setOpticalZoomFactorSync

MH2033, 2021-08-03:
After calling this API to set the zoom factor for the main camera the zoom factor is set and can even be seen from the radio controller but the setting is not applied unless i change the mode from wide to zoom from the controller and i couldn't find the correct API to do the same thing using the OSDK. Am i missing something here? Regards

dji-dev, 2021-08-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51223'>Zendesk ticket #51223</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

setOpticalZoomFactorSync only supports the Zoom lens. And OSDK provides API to change the camera lens:
  LiveViewErrCode changeH264Source(LiveViewCameraPosition pos, LiveViewCameraSource source);

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2021-08-04:
Thank you very much, this was the correct API to call
Best regards

closed 2021-08-04 by MH2033