dji-sdk/Onboard-SDK #818: H20相机码流GDR编码

github4529, 2021-08-04:
M300 + H20

由于H20/Z30的H264码流是GDR编码的，直接推送H264码流(rtmp)推流无法播放，现在基本都是OSDK的ffmpeg软解码，然后再调用编码（软/硬）为常规H264码流推送。

但是软解码占用CPU太高了，比如在树莓派上，花屏的现象会非常严重。

想问有没有什么解决方案。
硬解码仅在论坛里有NVIDIA平台的GPU有Gstreamer方案的硬解码，树莓派不支持。

无人机与遥控器的视频推送是如何解决GDR编码的呢，有没有相关的建议

dji-dev, 2021-08-04:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/51263'>Zendesk ticket #51263</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

github4529, 2021-08-04:
或者M300有没有适配的IDR的云台相机

dji-dev, 2021-08-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51263'>Zendesk ticket #51263</a>:<br><br>尊敬的开发者，

您好！感谢您联系DJI大疆创新。

抱歉，当前OSDK获取的视频流是GDR格式，遥控端涉及无线链路传输是IDR格式。因为硬解码与平台有关，可以尝试基于您自己使用的平台开发解码方案。

另后续关于SDK的问题，可以通过上面提供的表单链接提交，以便及时对您的问题进行跟进和确认。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

github4529, 2021-08-05:
intra-refresh
默认：off
说明：让x264为每keyint数量的帧使用宏块内部编码取代IDR帧。块以水平移动列的方式更新，也叫刷新波。对于低延迟的流，这样可以让帧的尺寸比使用标准的IDR帧更加保持恒定。而且这样可以增强视频流对丢包的容错能力。这个选项会降低压缩率，所以在确实需要的时候才选择它。
还有一些有意思的事情：1、第一帧依然是IDR帧。2、内部宏块只在P帧中存在，刷新波在一个或多个B帧后的P帧中广泛存在。3、主要的压缩率下降原因是在宏块中新（左边）的波并不能参考旧（右边）的波。
建议：使用默认值

生成的h264，是否是使用x264，有没有使用这个参数，将这个参数关闭的话是否就不是GDR编码的h264流了

dji-kyle, 2022-03-31:
OSDK从飞机相机拿到的流不可以更改，可以评估插 I 帧的方式解决解码问题

closed 2022-03-31 by dji-kyle