dji-sdk/Onboard-SDK #819: P1 getting bin and mrk files

deadman2000, 2021-08-06:
In the request for a list of files from the camera P1, only media is received. But we need both bin and mkr too. Is support for these files planned? Or is this a bug and will be fixed?

dji-dev, 2021-08-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51409'>Zendesk ticket #51409</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Do you develope OSDK application and use P1? OSDK has not yet adapted to P1, it will be adapted in the next version.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

dji-kyle, 2022-03-31:
PSDK 3.x has been released and adapted to P1

closed 2022-03-31 by dji-kyle