dji-sdk/Onboard-SDK #820: Image ISO Manifold 2-G

Caufmann, 2021-08-12:
Hi,

I'm currently working on manifold 2-G and I was looking for it's ISO Image.

Using this link : https://www.dji.com/fr/manifold-2/downloads , I can't find any ISO Image in the "Manifold 2 - Firmware V0.3.3.2 (Manifold 2-G Image Tool & Official Image)".gz file.

Could you give me some hints to find it, or provide it somehow ?

Looking forward to hearing from you,

Clément

dji-dev, 2021-08-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51679'>Zendesk ticket #51679</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Regarding the Manifold2 image file, it is recommended to directly consult the product after-sales service: support@dji.com

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

IuliuNovac, 2021-11-08:
You don't need an iso for manifold 2-G, it's not the c version. What you need is to follow DJI manifold 2-g documentation to unzip the gz archive and run the correct commands from a ubuntu machine to install the os of the manifold 2G. The manifold 2G is a jetson tx with a custom carrier, so you should treat it accordingly. Don't try to install it with the jetpack, because the DJI screwed up a here, instead of releasing the config file of the board pin configuration, they wrapped it into in the 3.3.2 version from jetpack and now you can't run the newest version for jetsons, but only the last ubuntu 16. I have tried to extract it, but due to lack of knowledge, I have failed.

closed 2022-03-31 by dji-kyle