dji-sdk/Onboard-SDK #821: Getting recording status and zoom mode

MH2033, 2021-08-15:
Hi,

OSDK provides APIs for changing the camera source(changeSourceH264) and video recording. But are there any APIs to get the recording status and zoom mode from the flight controller?(in case the user changed these using the radio controller)

Regards

dji-dev, 2021-08-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/51752'>Zendesk ticket #51752</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry，OSDK does not support to get the lens mode. but there is an API to get camera working mode:
  /*! @brief get camera working mode, blocking calls
  *
  *  @platforms M210V2, M300
  *  @note All the APIs whose name ending with sync or async in this class
  * have a restriction on calling. All these API should not be called until
  * the previous request receives ack or timeout.
  *  @param index camera module index, input limit see enum
  * DJI::OSDK::PayloadIndexType
  *  @param workingMode used as an output param, camera working mode, input
  * limit see enum DJI::OSDK::CameraModule::WorkMode
  *  @param timeout blocking timeout in seconds
  *  @return ErrorCode::ErrorCodeType error code
  */
  ErrorCode::ErrorCodeType getModeSync(PayloadIndexType index,
                                      CameraModule::WorkMode &workingMode,
                                      int timeout);

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

MH2033, 2021-08-22:
Thank you for your response

I've already worked with the mentioned API but this API only returns the working mode of the camera but it doesn't inform the caller if it's in the recording state or not

closed 2022-03-31 by dji-kyle