dji-sdk/Onboard-SDK #827: Is there a way to get thumbnails using OSDK?

akamilkhan, 2021-10-10:


dji-dev, 2021-10-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/54311'>Zendesk ticket #54311</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, OSDK currently does not support direct access to thumbnails. It provides API for obtaining camera video streams.
M300 supports downloading photo files from the camera (such as H20) SD card.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-10-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/54311'>Zendesk ticket #54311</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry,currently there is no relevant feature plan.
If you have requirements for relevant functionality, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://djisdksupport.zendesk.com/hc/en-us/requests/new

In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>