dji-sdk/Onboard-SDK #842: error: ‘av_warn_unused_result’ does not name a type  av_warn_unused_result

huynhtnhut97, 2021-11-04:
Hello,
I successfully compiled with cmake -DADVANCED_SENSING=ON . However, I encounter the following error when I command 'make djiosk-core'. 

`In file included from /home/civilcm/Gits/Onboard-SDK-4.1.0/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.hpp:15:0,
                 from /home/civilcm/Gits/Onboard-SDK-4.1.0/osdk-core/advanced-sensing/api/src/dji_advanced_sensing.cpp:24:
/usr/include/aarch64-linux-gnu/libavformat/avformat.h:2443:1: error: ‘av_warn_unused_result’ does not name a type
 av_warn_unused_result
 ^~~~~~~~~~~~~~~~~~~~~
/usr/include/aarch64-linux-gnu/libavformat/avformat.h:2465:1: error: ‘av_warn_unused_result’ does not name a type
 av_warn_unused_result
 ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/civilcm/Gits/Onboard-SDK-4.1.0/osdk-core/advanced-sensing/camera_stream/src/dji_camera_stream_decoder.hpp:16:0,
                 from /home/civilcm/Gits/Onboard-SDK-4.1.0/osdk-core/advanced-sensing/api/src/dji_advanced_sensing.cpp:24:
/usr/include/aarch64-linux-gnu/libswscale/swscale.h:156:1: error: ‘av_warn_unused_result’ does not name a type
 av_warn_unused_result
 ^~~~~~~~~~~~~~~~~~~~~
osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/build.make:62: recipe for target 'osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/api/src/dji_advanced_sensing.cpp.o' failed
make[3]: *** [osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/api/src/dji_advanced_sensing.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/all' failed
make[2]: *** [osdk-core/advanced-sensing/CMakeFiles/advanced-sensing.dir/all] Error 2
CMakeFiles/Makefile2:99: recipe for target 'osdk-core/CMakeFiles/djiosdk-core.dir/rule' failed
make[1]: *** [osdk-core/CMakeFiles/djiosdk-core.dir/rule] Error 2
Makefile:164: recipe for target 'djiosdk-core' failed
make: *** [djiosdk-core] Error 2
`

dji-dev, 2021-11-04:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/55743'>Zendesk ticket #55743</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

huynhtnhut97, 2021-11-04:
I solved it by commenting ‘av_warn_unused_result’ in corresponding files.

closed 2021-11-04 by huynhtnhut97