dji-sdk/Onboard-SDK #843: How to stay in mission for M300?

btsatoukzn, 2021-11-05:
hi, 

For M210, WaypointActionTypeFormat.WP_ACTION_STAY exists.
There is no equivalent in the M300.
For M300, what should I do to wait during the mission?

Can it be realized by combining SampleRearchPoint and AssociateAction of DJIWaypointV2ActionTriggerType?

I was using OSDK 4.1.0 and was able to run the sample program djiosdk-waypoint-v2-sample.

Thank you.

dji-dev, 2021-11-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/55840'>Zendesk ticket #55840</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Please refer to the link:
https://sdk-forum.dji.net/hc/en-us/articles/900005820706-How-to-set-waypoint-action-hover-in-waypoint-V2-

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2021-11-09:
thank you for your reply.

The contents of the above site do not have enough information on standby settings.

> DJIWaypointV2AircraftControlFlyingParam,   // stops flying --->A
> DJIWaypointV2IntervalTriggerParam,             // interval time --->B
> DJIWaypointV2AircraftControlFlyingParam,   // starts flying --->C

The action of M300 needs to set the trigger and the actuator.

A and C is actuator's parameter.
B is trigger's parameter.

What are the triggers for A and C?
What is B's actuator?

thank you.

btsatoukzn, 2021-11-09:
I tried the following settings.
However, the drone remained stationary after 10 seconds.

A. trigger: TriggerTypeSampleReachPoint
      actuator: AircraftControlFlyingParam (isStartFlying = 0)

B. trigger: TriggerTypeInterval (interval = 100)
      actuator: AircraftControlFlyingParam (isStartFlying = 0)

C. trigger: TriggerTypeActionAssociated
      actuator: AircraftControlFlyingParam (isStartFlying = 1)

thank you.

btsatoukzn, 2021-11-09:
I was able to confirm the operation of waiting for 10 seconds with the following settings.

A. trigger: TriggerTypeSampleReachPoint
actuator: AircraftControlFlyingParam (isStartFlying = 0)

B. trigger: TriggerTypeActionAssociated (waitingTime = 100)
actuator: AircraftControlFlyingParam (isStartFlying = 1)

Thank you.

closed 2021-11-09 by btsatoukzn