dji-sdk/Onboard-SDK #844: Sigificant deviation between gimbal real Yaw and telemetry Yaw

IuliuNovac, 2021-11-11:
Hi,

We have a M300 with the H20T camera, where we calibrated the RGB data from the images with the Telemetry obtained from the OSDK.
There is a significant deviation and inconsistency between the calibrated telemetry data and raw telemetry. The major issue is the rotation in Yaw.  We have conducted the calibration as follows:
We gathered telemetry data, then generated three sets of datasets for latitude, longitude, altitude , roll, pitch and Yaw. The only variation was the Yaw, where the first one is yaw of the gimbal, second is yaw of the gimbal minus drone yaw,third is the gimbal yaw plus drone yaw.
Afterwords we used the said datasets with identical images and calibrated in Pix4d.
The results show almost identical values for  latitude, longitude, altitude , roll, pitch and yaw for all three calibration.
But the issue is that they vary quite a bit from the input yaw, where gimbal yaw alone has the least standard deviation.

The data is combined in a excel file, where you can see the deviation. Moreover, there doesn't seem to be a deviation in time-step between the pictures and the telemetry
[CalibrationPix4dcomparison.xlsx](https://github.com/dji-sdk/Onboard-SDK/files/7519698/CalibrationPix4dcomparison.xlsx)
.

dji-dev, 2021-11-11:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/56188'>Zendesk ticket #56188</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

IuliuNovac, 2021-11-11:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/user/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/user/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

calib0 | Latc0 | lonc0 | altc0 | rolc0 | pitchc0 | yawc0 | calib1 | Latc1 | lonc1 | altc1 | rolc1 | pitchc1 | yawc1 | calib2 | Latc2 | lonc2 | altc2 | rolc2 | pitchc2 | yawc2 | raw0 | Lat0 | lon0 | alt0 | rol0 | pitch0 | yaw0 | raw1 | Lat1 | lon1 | alt1 | rol1 | pitch1 | yaw1 | raw2 | Lat2 | lon2 | alt2 | rol2 | pitch2 | yaw2
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
RGB1007.jpeg | 57.01525 | 9.979441 | 83.19483 | 3.469735 | 3.02088 | 45.13673 | RGB1007.jpeg | 9.979465 | 57.01526 | 82.85284 | 3.112465 | 4.127385 | 45.18661 | RGB1007.jpeg | 9.979465 | 57.01526 | 82.8481 | 3.12388 | 4.121718 | 45.20251 | RGB1007.jpeg | 57.01526 | 9.979489 | 83.94274 | 1.2 | 16.1 | -79.5 | RGB1007.jpeg | 57.01526 | 9.979489 | 83.94274 | 1.2 | 16.1 | -43.2186 | RGB1007.jpeg | 57.01526 | 9.979489 | 83.94274 | 1.2 | 16.1 | -115.781
RGB1008.jpeg | 57.01526 | 9.979451 | 82.48124 | -1.24622 | -3.03705 | 60.20428 | RGB1008.jpeg | 9.979487 | 57.01526 | 82.17652 | -1.59241 | -1.4118 | 60.32136 | RGB1008.jpeg | 9.979488 | 57.01526 | 82.12047 | -1.58143 | -1.3776 | 60.33097 | RGB1008.jpeg | 57.01527 | 9.97945 | 83.91584 | 0 | 0 | -4.1 | RGB1008.jpeg | 57.01527 | 9.97945 | 83.91584 | 0 | 0 | 30.35552 | RGB1008.jpeg | 57.01527 | 9.97945 | 83.91584 | 0 | 0 | -38.5555
RGB1009.jpeg | 57.01526 | 9.97946 | 82.37569 | -1.43465 | -2.44898 | 23.45913 | RGB1009.jpeg | 9.979489 | 57.01526 | 82.10945 | -1.79639 | -1.19766 | 23.54519 | RGB1009.jpeg | 9.979489 | 57.01526 | 82.10957 | -1.78806 | -1.19751 | 23.5491 | RGB1009.jpeg | 57.01527 | 9.979448 | 83.58052 | 0 | 0 | 1.6 | RGB1009.jpeg | 57.01527 | 9.979448 | 83.58052 | 0 | 0 | 39.10155 | RGB1009.jpeg | 57.01527 | 9.979448 | 83.58052 | 0 | 0 | -35.9016
RGB1010.jpeg | 57.01526 | 9.979438 | 82.1042 | -1.52944 | -3.26089 | 7.122839 | RGB1010.jpeg | 9.979437 | 57.01526 | 82.12688 | -1.49728 | -3.25919 | 7.107007 | RGB1010.jpeg | 9.979437 | 57.01526 | 82.12811 | -1.48953 | -3.25664 | 7.105726 | RGB1010.jpeg | 57.01527 | 9.979446 | 83.84723 | 0 | 0.099998 | 5.6 | RGB1010.jpeg | 57.01527 | 9.979446 | 83.84723 | 0 | 0.099998 | 43.61897 | RGB1010.jpeg | 57.01527 | 9.979446 | 83.84723 | 0 | 0.099998 | -32.419
RGB1011.jpeg | 57.01526 | 9.979435 | 82.08237 | -1.55473 | -3.3028 | 3.379333 | RGB1011.jpeg | 9.979435 | 57.01526 | 82.1049 | -1.54606 | -3.2671 | 3.371598 | RGB1011.jpeg | 9.979434 | 57.01526 | 82.10615 | -1.53454 | -3.27451 | 3.369555 | RGB1011.jpeg | 57.01527 | 9.979443 | 84.06027 | 0 | 0 | 6.4 | RGB1011.jpeg | 57.01527 | 9.979443 | 84.06027 | 0 | 0 | 41.48107 | RGB1011.jpeg | 57.01527 | 9.979443 | 84.06027 | 0 | 0 | -28.6811
RGB1012.jpeg | 57.01526 | 9.97944 | 82.08435 | -1.56372 | -3.05717 | 1.071888 | RGB1012.jpeg | 9.979439 | 57.01527 | 82.09855 | -1.54057 | -3.06057 | 1.054938 | RGB1012.jpeg | 9.979438 | 57.01527 | 82.09882 | -1.52262 | -3.06263 | 1.054834 | RGB1012.jpeg | 57.01527 | 9.97944 | 83.33598 | 0 | 0 | 6.4 | RGB1012.jpeg | 57.01527 | 9.97944 | 83.33598 | 0 | 0 | 34.96064 | RGB1012.jpeg | 57.01527 | 9.97944 | 83.33598 | 0 | 0 | -22.1606
RGB1013.jpeg | 57.01527 | 9.979438 | 82.07721 | -1.52793 | -3.00152 | 0.632723 | RGB1013.jpeg | 9.979438 | 57.01527 | 82.10114 | -1.50217 | -2.95522 | 0.609402 | RGB1013.jpeg | 9.979438 | 57.01527 | 82.10354 | -1.48853 | -2.95358 | 0.607023 | RGB1013.jpeg | 57.01528 | 9.979442 | 82.66036 | 0 | 0 | 7.2 | RGB1013.jpeg | 57.01528 | 9.979442 | 82.66036 | 0 | 0 | 31.89103 | RGB1013.jpeg | 57.01528 | 9.979442 | 82.66036 | 0 | 0 | -17.491
RGB1014.jpeg | 57.01527 | 9.979437 | 82.08464 | -1.51949 | -2.93827 | 0.645071 | RGB1014.jpeg | 9.979437 | 57.01527 | 82.11283 | -1.53819 | -2.90339 | 0.621368 | RGB1014.jpeg | 9.979437 | 57.01527 | 82.1131 | -1.52534 | -2.90228 | 0.620756 | RGB1014.jpeg | 57.01528 | 9.979447 | 82.84348 | 0 | 0.099998 | 10.4 | RGB1014.jpeg | 57.01528 | 9.979447 | 82.84348 | 0 | 0.099998 | 32.31342 | RGB1014.jpeg | 57.01528 | 9.979447 | 82.84348 | 0 | 0.099998 | -11.5134
RGB1015.jpeg | 57.01527 | 9.979433 | 82.08492 | -1.63808 | -3.10565 | -0.61652 | RGB1015.jpeg | 9.979431 | 57.01527 | 82.09889 | -1.57799 | -3.17037 | -0.63365 | RGB1015.jpeg | 9.979431 | 57.01527 | 82.09954 | -1.56384 | -3.16801 | -0.63616 | RGB1015.jpeg | 57.01528 | 9.979454 | 82.66222 | 0 | 0.099998 | 13.4 | RGB1015.jpeg | 57.01528 | 9.979454 | 82.66222 | 0 | 0.099998 | 32.88632 | RGB1015.jpeg | 57.01528 | 9.979454 | 82.66222 | 0 | 0.099998 | -6.08632
RGB1016.jpeg | 57.01527 | 9.979436 | 82.10596 | -1.64566 | -3.13921 | -3.35415 | RGB1016.jpeg | 9.979436 | 57.01527 | 82.13299 | -1.65809 | -3.11408 | -3.37609 | RGB1016.jpeg | 9.979435 | 57.01527 | 82.13343 | -1.64488 | -3.12331 | -3.37575 | RGB1016.jpeg | 57.01529 | 9.979462 | 83.23876 | 0 | 0.099998 | 15.9 | RGB1016.jpeg | 57.01529 | 9.979462 | 83.23876 | 0 | 0.099998 | 33.15141 | RGB1016.jpeg | 57.015 [... cut ...] | 9.979936 | 82.77862 | 0 | 0.099998 | 10.28388
RGB1042.jpeg | 57.01529 | 9.979933 | 82.72506 | -1.432 | -1.26613 | 7.830324 | RGB1042.jpeg | 9.979933 | 57.01529 | 82.73797 | -1.3919 | -1.26122 | 7.814042 | RGB1042.jpeg | 9.979933 | 57.01529 | 82.73943 | -1.37414 | -1.26067 | 7.813062 | RGB1042.jpeg | 57.01529 | 9.979937 | 83.27402 | 0 | 0 | -0.2 | RGB1042.jpeg | 57.01529 | 9.979937 | 83.27402 | 0 | 0 | -11.7784 | RGB1042.jpeg | 57.01529 | 9.979937 | 83.27402 | 0 | 0 | 11.37839

</body>

</html>

IuliuNovac, 2021-11-11:
[Calibration_comparison_2.xlsx](https://github.com/dji-sdk/Onboard-SDK/files/7521996/Calibration_comparison_2.xlsx)
This is the second test, here the yaw was mirrored by (yaw*-1). Before the data was mesured the magnetic, imu sensors and gimbal where re calibrated