dji-sdk/Onboard-SDK #846: Compile single examples without errors

highfreq, 2021-11-20:
Followed the tutorial but i still get errors when trying to compile single examples (entire project compiles with no prob).

Would it be possible to add a tutorial page to dev with steps needed to compile single examples.

There is a similar problem here on github with a solution, i tried that CMakeFile.txt but i still get dji_log.hpp error.

So it would be of great elp if devs give a turorial on how to compile a single example so that people could start coding and not worry about linking and placing stuff here and there.

Thanks

dji-dev, 2021-11-20:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/56647'>Zendesk ticket #56647</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

highfreq, 2021-11-23:
Tested you CMakelists.txt for telemetry and it works, do you know what adjustments should be done to have it working for flight-control ??
If i use same CMakeLists.txt as the one used to build telemetry it fails at the end when linking CXX executable.
I am trying to understand what each needs.

Scanning dependencies of target djiosdk-flightcontrol-sample
[ 14%] Building CXX object CMakeFiles/djiosdk-flightcontrol-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_environment.cpp.o
[ 28%] Building CXX object CMakeFiles/djiosdk-flightcontrol-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o
[ 42%] Building C object CMakeFiles/djiosdk-flightcontrol-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/hal/osdkhal_linux.c.o
[ 57%] Building C object CMakeFiles/djiosdk-flightcontrol-sample.dir/home/pi/Onboard-SDK/sample/platform/linux/osal/osdkosal_linux.c.o
[ 71%] Building CXX object CMakeFiles/djiosdk-flightcontrol-sample.dir/flight_control_sample.cpp.o
[ 85%] Building CXX object CMakeFiles/djiosdk-flightcontrol-sample.dir/main.cpp.o
[100%] Linking CXX executable djiosdk-flightcontrol-sample
/usr/bin/ld: CMakeFiles/djiosdk-flightcontrol-sample.dir/main.cpp.o: in function main': /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:72: undefined reference to FlightSample::FlightSample(DJI::OSDK::Vehicle*)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:95: undefined reference to FlightSample::monitoredTakeoff(int)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:96: undefined reference to FlightSample::monitoredLanding(int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:101: undefined reference to FlightSample::monitoredTakeoff(int)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:104: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:106: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:108: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:110: undefined reference to FlightSample::monitoredLanding(int)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:118: undefined reference to FlightSample::monitoredTakeoff(int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:123: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:126: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:129: undefined reference to FlightSample::setNewHomeLocation(int)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:132: undefined reference to FlightSample::setGoHomeAltitude(unsigned short, int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:135: undefined reference to FlightSample::moveByPositionOffset(DJI::OSDK::Telemetry::Vector3f const&, float, float, float)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:140: undefined reference to FlightSample::goHomeAndConfirmLanding(int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:149: undefined reference to FlightSample::monitoredTakeoff(int)' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:152: undefined reference to FlightSample::velocityAndYawRateCtrl(DJI::OSDK::Telemetry::Vector3f const&, float, unsigned int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:154: undefined reference to FlightSample::emergencyBrake()' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:156: undefined reference to FlightSample::velocityAndYawRateCtrl(DJI::OSDK::Telemetry::Vector3f const&, float, unsigned int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:158: undefined reference to FlightSample::emergencyBrake()' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:160: undefined reference to FlightSample::velocityAndYawRateCtrl(DJI::OSDK::Telemetry::Vector3f const&, float, unsigned int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:162: undefined reference to FlightSample::emergencyBrake()' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:164: undefined reference to FlightSample::velocityAndYawRateCtrl(DJI::OSDK::Telemetry::Vector3f const&, float, unsigned int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:166: undefined reference to FlightSample::emergencyBrake()' /usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:169: undefined reference to FlightSample::monitoredLanding(int)'
/usr/bin/ld: /home/pi/Onboard-SDK/sample/platform/linux/flight-control/main.cpp:178: undefined reference to `FlightSample::~FlightSample()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/djiosdk-flightcontrol-sample.dir/build.make:183: djiosdk-flightcontrol-sample] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/djiosdk-flightcontrol-sample.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

A bunch of undefined references to FlightSample.