dji-sdk/Onboard-SDK #847: OSDK 4.1 request file list broken with firmware 03.00.0101

vdebad, 2021-11-23:
I am trying to download image files from the Zenmuse H20T's SD-Card for image processing. When requesting file list, the call returns with the error "parseFileList, L710: File list total data len error !". See code at the end of this post for reproduction.

Firmware version:
Aircraft: 03.00.0101
Remote control: 03.00.0102
Battery1: 01.02.05.31
Battery2: 01.02.05.31
Zenmuse H20T: 03.00.01.02

Here is the log:
```
Read App ID
User Configuration read successfully. 

[1288463.414]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1288463.414]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9G00C128M
[1288463.414]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.93
[1288463.414]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[1288463.414]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[1288464.414]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[1288464.614]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[1288464.614]STATUS/1 @ FileMgrImpl, L308: register download file callback handler successfully.
[1288464.637]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[1288464.637]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[1288465.638]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[1288465.648]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[1288465.648]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[1288465.650]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1288465.650]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[1288465.652]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[1288465.652]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[1288465.654]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[1288465.654]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[1288466.638]STATUS/1 @ firewallTask, L112: firewall task created ...
[1288466.663]STATUS/1 @ getDroneVersion, L1702: ret = 0
[1288466.663]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9G00C128M
[1288466.663]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.93
[1288466.669]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[1288466.669]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[1288466.670]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[1288466.670]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[1288466.670]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[1288466.670]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[1288467.470]STATUS/1 @ init, L254: Start advanced sensing initalization
[1288467.470]STATUS/1 @ activate, L1329: version 0x304085D

[1288467.493]STATUS/1 @ activate, L1367: Activation successful

Camera module name: H20T.C
[1288468.495]STATUS/1 @ main, L212: Getting work mode of camera at payload index 0 (H20T)
[1288468.537]STATUS/1 @ main, L224: Current work mode:
[1288468.537]STATUS/1 @ main, L228: Capture mode. In this mode, the user can capture pictures.
[1288468.537]STATUS/1 @ main, L265: Getting shoot photo mode of camera at payload index 0 (H20T)
[1288468.557]ERRORLOG/1 @ main, L274: Could not get camera's shoot-photo mode using 'getShootPhotoModeSync'. Error code: 0xE0
[1288468.557]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Camera
[1288468.557]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Command not supported
[1288468.557]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Check the firmware or command validity
[1288468.557]STATUS/1 @ main, L312: Taking single picture with camera at payload index 0 (H20T)
[1288470.577]STATUS/1 @ main, L333: Setting camera's work mode to PLAYBACK
[1288470.598]ERRORLOG/1 @ main, L340: Could not set camera's work mode to playback using 'setModeSync(DJI::OSDK::PAYLOAD_INDEX_0, DJI::OSDK::CameraModule::WorkMode::PLAYBACK, 2)'. Error code: 0xD9
[1288470.598]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Camera
[1288470.598]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Unknown
[1288470.598]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : ����
[1288470.598]STATUS/1 @ main, L349: Obtain read permission to camera memory from camera, blocking calls
[1288470.677]STATUS/1 @ fileListMonitorTask, L154: OSDK download monitor task created.
[1288470.768]STATUS/1 @ OnReceiveDataPack, L1076: [First pack] get the first pack
[1288470.769]STATUS/1 @ parseFileList, L706: ###data->amount = 24, data->len = 6560
[1288470.769]ERRORLOG/1 @ parseFileList, L710: File list total data len error !
[1288470.769]STATUS/1 @ SendAbortPack, L1137: SendAbortPack
[1288470.769]STATUS/1 @ fileMgrFileListCb, L125: Unknown file type.
[... 361 lines cut ...]
				{
					DERROR(
						"Could not obtain read permission for camera storage. "
						"Error code: 0x%1X",
						ret);
					ErrorCode::printErrorCodeMsg(ret);
				}

				ret = cm->startReqFileList(DJI::OSDK::PAYLOAD_INDEX_0,
										   fileMgrFileListCb, NULL);

				sleep(5);  // wait 5 seconds for callback

				if (ret != ErrorCode::SysCommonErr::Success)
				{
					DERROR("Could not get file list. Error code: 0x%1X", ret);
					ErrorCode::printErrorCodeMsg(ret);
				}
			}
		}
	}

	return 0;
}
```

dji-dev, 2021-11-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/56781'>Zendesk ticket #56781</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the inconvenience. We have noticed this problem. After upgrading the new firmware, download only works with less than 3 photos in the SD card. We have submitted this bug and we will fix it later. If this function blocks your development, it is recommended to temporarily roll back the version.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

vdebad, 2021-11-24:
A temporary fix is to comment checking of frame-length in `dji_file_mgr_impl.cpp`in function `parseFileList` line 596 to 599:

Comment this code block:
```
  if (listdata->len > fullDataList.size() * maxDataFrameLen) {
    DERROR("File list total data len error !");
    return pack;
  }
```

Please beware: I know file listing is working afterwards, but I cannot guarantee, that this fix does not break any other features, so be careful!