dji-sdk/Onboard-SDK #849: FlightController wind compensation

deadman2000, 2021-12-01:
Hi, DJI!

We are trying to implement an automatic drone landing. But at the moment we have a great chance to crash the drone. Because during OSDK control, the drone ignores the wind. This means that we don't know what value to pass to the flightController so that the drone will fly to where we need it. The STABLE_ENABLE flag does not help solve this problem.

dji-dev, 2021-12-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57198'>Zendesk ticket #57198</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

When landing automatically, what mode does the OSDK flightCtrl use in the vertical? If the throttle control mode is used, the drone may not be stabilized. You can consider obtaining the drone quaternion and feedback to increase the stability.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

deadman2000, 2021-12-06:
We are using these modes: VERTICAL_VELOCITY | HORIZONTAL_VELOCITY | HORIZONTAL_BODY | YAW_RATE | STABLE_ENABLE

Can you explain how to use quaternion to hold the position?