dji-sdk/Onboard-SDK #850: error: ‘CV_BGR2GRAY’ was not declared in this scope

fatbringer, 2021-12-02:
After running succesfully
```
cmake .. -DADVANCED_SENSING=ON -DTARGET_TRACKING_SAMPLE=ON 
````
I am unable to perform sudo make -j7 install
Error returned is
```
sudo make -j7 install
[ 13%] Built target advanced-sensing
[ 32%] Built target djiosdk-core
[ 35%] Built target djiosdk-waypoint-v2-sample
[ 38%] Built target djiosdk-mission-sample
[ 40%] Built target djiosdk-mfio-sample
[ 43%] Built target djiosdk-flightcontrol-sample
[ 45%] Built target djiosdk-telemetry-sample
[ 48%] Built target djiosdk-logging-sample
[ 51%] Built target djiosdk-mobile-sample
[ 54%] Built target time_sync_poll_sample
[ 59%] Built target payloads_async_sample
[ 61%] Built target time_sync_callback_sample
[ 63%] Built target stereo-vision-single-thread-sample
[ 68%] Built target payloads_sync_sample
[ 71%] Built target djiosdk-payload-control-sample
[ 75%] Built target payloads_download_sample
[ 77%] Built target stereo-vision-new-sample
[ 79%] Built target camera-stream-poll-sample
[ 82%] Built target stereo-vision-multi-thread-sample
[ 84%] Built target camera-stream-callback-sample
[ 84%] Building CXX object sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/KCFcpp/src/kcftracker.cpp.o
[ 86%] Built target djiosdk-hms-sample
[ 88%] Built target djiosdk-liveview-sample
[ 90%] Built target op_upload_sample
[ 92%] Built target djiosdk-battery-sample
[ 94%] Built target om_download_sample
[ 96%] Built target op_download_sample
In file included from /home/dji/Desktop/djiOSDK/src/Onboard-SDK-4.1.0/sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/KCFcpp/src/kcftracker.cpp:86:0:
/home/dji/Desktop/djiOSDK/src/Onboard-SDK-4.1.0/sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/KCFcpp/src/recttools.hpp: In function ‘cv::Mat RectTools::getGrayImage(cv::Mat)’:
/home/dji/Desktop/djiOSDK/src/Onboard-SDK-4.1.0/sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/KCFcpp/src/recttools.hpp:132:28: error: ‘CV_BGR2GRAY’ was not declared in this scope
     cv::cvtColor(img, img, CV_BGR2GRAY);
                            ^~~~~~~~~~~
sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/build.make:86: recipe for target 'sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/KCFcpp/src/kcftracker.cpp.o' failed
make[2]: *** [sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/KCFcpp/src/kcftracker.cpp.o] Error 1
CMakeFiles/Makefile2:1350: recipe for target 'sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/all' failed
make[1]: *** [sample/platform/linux/advanced-sensing/camera_stream_target_tracking_sample/CMakeFiles/kcf_tracker.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

````

Previously I have not encounter this issue with OSDK 4.1.0
I have always use OSDK 4.1.0. 
Opencv version
```
dpkg -l | grep libopencv
ii  libopencv                                     4.1.1-2-gd5a58aa75                               arm64        Open Computer Vision Library
ii  libopencv-dev                                 4.1.1-2-gd5a58aa75                               arm64        Development files for Open Source Computer Vision Library
ii  libopencv-python                              4.1.1-2-gd5a58aa75                               arm64        Python bindings for Open Source Computer Vision Library
ii  libopencv-samples                             4.1.1-2-gd5a58aa75                               arm64        Samples for Open Source Computer Vision Library
````

dji-dev, 2021-12-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57262'>Zendesk ticket #57262</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

OSDK sample third-party libs, please refer to the following link:
https://sdk-forum.dji.net/hc/en-us/articles/900006035306-The-OSDK-sample-compile-error-reports-that-the-lib-and-header-files-cannot-be-found-

OSDK  sample based on opencv 3.3.1 and opencv-contrib 3.3.1

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57262'>Zendesk ticket #57262</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry，OSDK 4.1 will stop updating new functions and will not adapt to opencv 4 in the future.
In fact, opencv is used for the sample functions, and developers can adapt opencv 4.x to develop OSDK applications.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

highfreq, 2021-12-10:
Really sad a company this big has so poor support for devs. What trouble could it be to adapt sample to OpenCV 4...............really really sad DJI cares so little about devs.
It look like DJI doesn't really like people to dev on their products.

fatbringer, 2022-03-10:
Hi @dji-dev for OSDK, is it possible to run the OSDK and OSDK ROS on other embedded computer platform such as NVIDIA AGX Xavier ?  Instead of NVIDIA jetson tx2 like the DJI Manifold 2G ?

SilaLiu, 2023-05-18:
修改方法为：将CV_BGR2GRAY改为cv::COLOR_BGR2GRAY