dji-sdk/Onboard-SDK #851: Error linking during compilation with Advanced Sensing

Caseiber, 2021-12-03:
I am trying to compile a file that imports the OSDK and uses advanced sensing. Everything seems to build, but during linking I get an error. This is my CMake file.

```
cmake_minimum_required(VERSION 3.0)
project(camera_access)

# Set compiler flags + our source files
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread -g -O0")
set(THREADS_PREFER_PTHREAD_FLAG ON)
set(SRC_FILES ${SRC_FILES} camera_access.cpp)

# Enable advanced sensing
set(ADVANCED_SENSING 1)
if (ADVANCED_SENSING)
  add_definitions(-DADVANCED_SENSING)
endif()

# CPU-specific stuff
if(CMAKE_SYSTEM_PROCESSOR MATCHES "i386|i686|x86|AMD64|x86_64")
   set(ARCH x86)
   add_definitions(-DDJIOSDK_HARDWARE_TYPE=3)
elseif((CMAKE_SYSTEM_PROCESSOR MATCHES "^arm*") OR (CMAKE_SYSTEM_PROCESSOR MATCHES "^aarch"))
  if(CMAKE_SIZEOF_VOID_P EQUAL 4)
     set(ARCH armv7)
     add_definitions(-DDJIOSDK_HARDWARE_TYPE=1)
  elseif(CMAKE_SIZEOF_VOID_P EQUAL 8)
     set(ARCH armv8)
     add_definitions(-DDJIOSDK_HARDWARE_TYPE=2)
  endif()
elseif()
  add_definitions(-DDJIOSDK_HARDWARE_TYPE=0)
endif()

# Set the DJI sources
if (NOT ONBOARDSDK_SOURCE)
    set(ONBOARDSDK_SOURCE "~/Projects/Onboard-SDK/osdk-core")
endif()
set(ORI_OSDK_CORE_SRC "${ONBOARDSDK_SOURCE}/advanced-sensing/ori-osdk-core")

# Include DJI header directories
include_directories(${ONBOARDSDK_SOURCE}/api/inc)
include_directories(${ONBOARDSDK_SOURCE}/utility/inc)
include_directories(${ONBOARDSDK_SOURCE}/hal/inc)
include_directories(${ONBOARDSDK_SOURCE}/protocol/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/linux/inc)
include_directories(${ONBOARDSDK_SOURCE}/platform/${ARCH}/inc)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/hal)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/hal/hotplug)
include_directories(${ONBOARDSDK_SOURCE}/../sample/platform/linux/osal)
include_directories(${ONBOARDSDK_SOURCE}/../sample/sore/inc)
include_directories(${ONBOARDSDK_SOURCE}/..sample/platform/linux/hal)
include_directories(${ONBOARDSDK_SOURCE}/..sample/platform/linux/hal/hotplug)
include_directories(${ONBOARDSDK_SOURCE}/..sample/platform/linux/osal)
include_directories(${ONBOARDSDK_SOURCE}/..sample/sore/inc)
include_directories(include /usr/local/include)
include_directories(include /usr/local/lib)

# Find OpenCV
find_package(OpenCV)
if (OpenCV_FOUND)
  message("\n${PROJECT_NAME}...")
  message(STATUS "Found OpenCV installed in the system, will use it to display image in AdvancedSensing APIs")
  message(STATUS " - Includes: ${OpenCV_INCLUDE_DIRS}")
  message(STATUS " - Libraries: ${OpenCV_LIBRARIES}")
  add_definitions(-DOPEN_CV_INSTALLED)
else()
  message( STATUS "Did not find OpenCV in the system, image data is inside RecvContainer as raw data")
endif ()

# Find LibUSB
[... 105 lines cut ...]
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x48): undefined reference to `OsdkLinux_TaskDestroy'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x50): undefined reference to `OsdkLinux_TaskSleepMs'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x58): undefined reference to `OsdkLinux_MutexCreate'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x60): undefined reference to `OsdkLinux_MutexDestroy'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x68): undefined reference to `OsdkLinux_MutexLock'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x70): undefined reference to `OsdkLinux_MutexUnlock'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x78): undefined reference to `OsdkLinux_SemaphoreCreate'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x80): undefined reference to `OsdkLinux_SemaphoreDestroy'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x88): undefined reference to `OsdkLinux_SemaphoreWait'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x90): undefined reference to `OsdkLinux_SemaphoreTimedWait'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0x98): undefined reference to `OsdkLinux_SemaphorePost'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0xa0): undefined reference to `OsdkLinux_GetTimeMs'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0xa8): undefined reference to `OsdkLinux_Malloc'
CMakeFiles/camera_access.dir/home/caroline/Projects/Onboard-SDK/sample/platform/linux/common/dji_linux_helpers.cpp.o:(.data.rel+0xb0): undefined reference to `OsdkLinux_Free'
collect2: error: ld returned 1 exit status
CMakeFiles/camera_access.dir/build.make:189: recipe for target 'camera_access' failed
make[2]: *** [camera_access] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/camera_access.dir/all' failed
make[1]: *** [CMakeFiles/camera_access.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
As you can see in my Cmake file, I am already importing both osdkhal_linux.c and osdkosal_linux.c.

Thank you.

dji-dev, 2021-12-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57351'>Zendesk ticket #57351</a>:<br><br>This issue seems same as the github issue, which maybe give you some referrcence.
https://github.com/dji-sdk/Onboard-SDK/issues/770
<br><p><em><small>°°°</small></em></p>

Caseiber, 2021-12-06:
Thank you for your reply. I am aware of that post and had gone through it to try and solve my issue. However, that person’s problem was resolved by importing these files, specifically the last two:

```
FILE(GLOB SOURCE_FILES *.hpp *.cpp
         ${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_environment.cpp
         ${CMAKE_CURRENT_SOURCE_DIR}/../common/dji_linux_helpers.cpp
         ${CMAKE_CURRENT_SOURCE_DIR}/../hal/osdkhal_linux.c
         ${CMAKE_CURRENT_SOURCE_DIR}/../osal/osdkosal_linux.c

         )
```

I am already importing these files and still having the same issue.

Thank you