dji-sdk/Onboard-SDK #852: Gimbal heading and drone headings can have large offsets when the gimbal has been reset

Gugfann, 2021-12-07:
Hello, we have been having issues with the gimbal yaw axis not centering properly when using gimbal angle commands. If you are trying to point the gimbal straight ahead by using the drone heading, it can be off by 10s of degrees in bad circumstances. But if you reset the yaw axis it will center as expected. You can actually see the offset on the remote controller. The little yellow tickmark that shows the gimbal heading also has the same offset as the one we read through the OSDK. It does not point straight ahead as expected after you have reset the gimbal.

There is almost always a small offset of 0-5 degrees, but we have seen more than 40 degrees in extreme cases. This usually correlates to situations where the drone is started in bad magnetic environments (inside, close to metal structures, and so on). 

This appears to be due to some underlying issue with how the gimbal yaw encoders are mapped to the drone heading. This could also mean that it is a problem that goes beyond the OSDK, but I'm posting it here because the OSDK seems to rely on the gimbal heading and drone heading being in the same reference frame. This is definitely not always the case since the drone reference frame changes with the magnetic environment, but the gimbal reference frame seems to be constant.

We have bypassed the issue somewhat by manually resetting the gimbal during takeoff and calculating the offset between the gimbal heading and the drone heading. But it would be great if you could work out a fix for this. Ideally, I would like an interface that was relative to the drone, like it used to be.

dji-dev, 2021-12-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57491'>Zendesk ticket #57491</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue seems to be related to the gimbal mode, OSDK 4.1 only supports gimbal yaw free mode, there may be errors between drone head and gimbal YAW, the newly integrated OSDK functions PSDK 3.0 already provides gimbal follow mode (follow drone head YAW).
https://github.com/dji-sdk/Payload-SDK/blob/b1f481e0d445268e2bd8ae5017d245db921faaeb/psdk_lib/include/dji_gimbal_manager.h#L71

Note:
PSDK 3.0 API does not compatible with OSDK 4.1.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

WatsonMing, 2023-02-24:
unbelievable