dji-sdk/Onboard-SDK #854: Is there a way to communicate to a PC on the ground?

tubzby, 2021-12-09:
Hi Dji,  I'm wondering if there is any way I can connect to a TCP server on the ground while I'm in air on the drone?

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57588'>Zendesk ticket #57588</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

There are two suggestions for referrence:
1. The ground end is communicate with the drone through MSDK (RC), MSDK connects to the server

2. Using onboard computer + OSDK development on the drone side. The onboard computer integrates wireless communication modules and develops communication with the ground server.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

tubzby, 2021-12-09:
Thanks for the quick reply and helpful suggestion, using mobile SDK might be a good option.

Just for clarification,  is that "wireless communication modules" in option 2 mean something like WiFi/4G/5G modules, are these kinds of modules available from Dji that I can just Plug and Play?

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57588'>Zendesk ticket #57588</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Just for clarification, is that "wireless communication modules" in option 2 mean something like WiFi/4G/5G modules, are these kinds of modules available from Dji that I can just Plug and Play?
    -->Yes, "wireless communication modules" in option 2 mean something like WiFi/4G/5G modules, DJI does not provide onboard computer wireless modules, developers need to choose and develop according to actual needs.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2021-12-09 by tubzby