dji-sdk/Onboard-SDK #855: Mop Connection issue

choiwooseok, 2021-12-09:
Problem

* PSDK app start -> negotiation FIN -> Started 
* OSDK <-> MSDK Mop pipeline Connection established

If the above time is almost the same, the OSDK <-> MSDK connection is disconnected. 

And then we cannot recovery the OSDK <-> MSDK connection..

[issue-660](https://github.com/dji-sdk/Onboard-SDK/issues/660)

Please give me help! Thanks.

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

- PSDK app start -> negotiation FIN -> Started
- OSDK <-> MSDK Mop pipeline Connection established
If the above time is almost the same, the OSDK <-> MSDK connection is disconnected.
And then we cannot recovery the OSDK <-> MSDK connection..
  -->Sorry, I can't understand this problem. Is that the PSDK start caused the OSDK <-->MSDK disconnected ? Whether the PSDK has started MOP and whether the pipeline ID conflicts. If it is covient, please provide the complete terminal logs. and operation steps.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-09:
```
MSDK(v4.15.1) <------ MOP Pipeline -----> OSDK(v4.1)
                                          PSDK(v3.0) for camera streaming.

PSDK application (Process 1)               Onboard SDK application (Process 2)
PSDK negotiation start                          loop until MSDK <-> OSDK connection establish
PSDK negotiation fin                             
```
Situation

If the time when the MSDK and OSDK connection establish time and the PSDK negotiation end time are almost the same, the MSDK and OSDK connection is immediately disconnected. 

And cannot recovery connection between MSDK <-> OSDK.

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Could you please provide the PSDK 3.0 payload hardware connections photos, and PSDK 3.0 and OSDK 4.1 complete terminal log ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-13:
[image]

dji-dev, 2021-12-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I test this issue with the steps:
1、PSDK 3.0 demo connects to the upward gimbal(running the demo: dji_sdk_demo_linux)
2、OSDK 4.1 connects to the OSDK expand port
3、wait the PSDK 3.0 application "waiting payload negotiate finished" and then connects MSDK MOP to the OSDK 4.1, it works fine.

PSDK terminal:
 [image]

OSDK terminal log(not the original demo code, changed for looping accepting)​
 [image]​

I would like to ask whether the above steps are correct. Please add the detailed steps. I will test it again.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/8ILxxn6XbrY7HN9ZbQtyCrp0p' target='zd_attach'>inline-1712902157.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/l1njiUA1EXh4nzCjvnYAKWRcW' target='zd_attach'>inline-363116837.png</a>
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-13:
1, 2 OK
> wait the PSDK 3.0 application "waiting payload negotiate finished" and then connects MSDK MOP to the OSDK 4.1, it works fine.

MSDK <-> OSDK 4.1 : Wait for accept success. At the same time wait PSDK payload negotiation finished.

then problem occurs MSDK <-> OSDK connection closed.

I will attach teminal log

choiwooseok, 2021-12-14:
First issue that I claimed. I will attach teminal log soon.

Second scenario

MSDK <-> OSDK Normal but PSDK got error.

```
[2021-12-13 17:26:55.249] [driver] [info] [FcIODjiImpl] MOP send Pipeline accept success
[2021-12-13 17:26:55.249] [driver] [info] [FcIODjiImpl] send data length : 282
[2021-12-13 17:26:55.724] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 282
[2021-12-13 17:26:56.129] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:26:56.231] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:26:56.300] [driver] [info] [FcIODjiImpl] MOP RECV Pipeline accept success
[2021-12-13 17:26:57.184] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:26:57.248] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:26:57.575] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:26:57.620] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:26:58.388] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:26:58.425] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:26:58.526] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:26:58.569] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:26:59.330] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:26:59.377] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:26:59.579] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:26:59.653] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:00.369] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:00.417] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:00.667] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:00.735] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:01.286] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:01.323] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:01.724] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:01.769] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:02.374] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:02.450] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:02.801] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:02.885] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:03.345] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:03.377] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:03.829] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:03.899] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:04.308] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:04.342] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:04.905] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:04.957] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:05.360] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:05.400] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:05.964] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:05.998] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:06.351] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:06.398] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:07.106] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:07.152] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:07.352] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:07.415] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88
[2021-12-13 17:27:08.116] [driver] [info] [FcIODjiImpl] send data length : 141
[2021-12-13 17:27:08.163] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 141
[2021-12-13 17:27:08.363] [driver] [info] [FcIODjiImpl] send data length : 88
[2021-12-13 17:27:08.448] [driver] [info] [FcIODjiImpl] send data by dji pipeline success. sent length : 88

```

PSDK die

```
[0.000][core]-[Info]-[DjiCore_Init:89) Payload SDK Version : V3.0.0-beta.0-build.1433
[0.002][utils]-[Debug]-[DjiCore_RootTask:131) work execution time:
[0.002][utils]-[Debug]-[DjiCore_RootTask:132) name	                execution time (ms)	  max execution time (ms)
[0.002][utils]-[Debug]-[DjiCore_RootTask:134) recorderWork                    	0       	0       
[0.002][utils]-[Debug]-[DjiCore_RootTask:134) flowControllerTask              	0       	0       
[... 828 lines cut ...]
[... cut ...]
[27.639][linker]-[Error]-[DjiLinker_RecvTask:261) send msg to queue error
[27.639][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C37
[27.649][utils]-[Error]-[DjiMsgq_Send:230) semaphore wait timeout
[27.649][linker]-[Error]-[DjiLinker_RecvTask:261) send msg to queue error
[27.871][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0xCA->0x01 0x0001
[27.882][utils]-[Error]-[DjiMsgq_Send:230) semaphore wait timeout
[27.882][linker]-[Error]-[DjiLinker_RecvTask:261) send msg to queue error
```

choiwooseok, 2021-12-14:
First issue that I claimed. Terminal Log

OSDK
```
[2021-12-14 16:08:36.194] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.195] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.195] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.196] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.196] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.198] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.198] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.199] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.199] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.199] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.200] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.200] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.200] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.201] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.201] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.201] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.201] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.201] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.202] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.202] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.202] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.202] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.203] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.203] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.210] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.210] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.211] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.212] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.212] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.212] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:36.212] [driver] [error] [FcIODjiImpl] MOP send Pipeline accept fail 5
[2021-12-14 16:08:40.295] [driver] [info] [FcIODjiImpl] MOP send Pipeline accept success
[2021-12-14 16:08:40.296] [driver] [info] [FcIODjiImpl] send data length : 423
[2021-12-14 16:08:40.299] [driver] [info] [FcIODjiImpl] MOP RECV Pipeline accept success
[2021-12-14 16:08:40.353] [driver] [info] [FcIODjiImpl] send pipeline disconnected
[2021-12-14 16:08:40.436] [driver] [info] [FcIODjiImpl] recv pipeline disconnected
```

PSDK
```
[0.000][core]-[Info]-[DjiCore_Init:89) Payload SDK Version : V3.0.0-beta.0-build.1440
[0.001][utils]-[Debug]-[DjiCore_RootTask:131) work execution time:
[0.001][utils]-[Debug]-[DjiCore_RootTask:132) name	                execution time (ms)	  max execution time (ms)
[0.001][utils]-[Debug]-[DjiCore_RootTask:134) recorderWork                    	0       	0       
[0.001][utils]-[Debug]-[DjiCore_RootTask:134) flowControllerTask              	0       	0       
[0.005][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.005][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.025][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.045][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.065][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.086][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.109][linker]-[Debug]-[DjiLinker_DeInit:164) deinit linker success, protocol:2, channelId:0
[0.111][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.111][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.131][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.152][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.172][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.192][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[... 760 lines cut ...]
[... cut ...]
[23.706][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C43
[23.706][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C37
[24.206][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C36
[24.206][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C43
[24.206][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C37
[24.702][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0x04 0x3C10
[24.706][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C36
```

dji-dev, 2021-12-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log record looks like OSDK sent a pipeline disconnected. It seems to be the log you added. Can you check where the code recorded this log?
 [image]
Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team​<br><a href='https://djisdksupport.zendesk.com/attachments/token/vQ4sGCsVmOJmFw2GPHHQeukQ6' target='zd_attach'>inline368038474.png</a>
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-14:
> FYI

void FcIODjiImpl::RecvThread() {
  uint8_t recvBuff[1024];

  while (true) {
    MopPipeline::DataPackType readPack = {recvBuff, 1024};

    if (mopPipelineRecv_ == nullptr) {
      MopErrCode acceptRet = vehicle_->mopServer->accept(MO_PIPELINE_RECV_ID, RELIABLE, mopPipelineRecv_);
      if (acceptRet != MOP_PASSED) {
        LoggerManager::GetLogger("driver")->error("[FcIODjiImpl] MOP RECV Pipeline accept fail {}", acceptRet);
        continue;
      } else {
        LoggerManager::GetLogger("driver")->info("[FcIODjiImpl] MOP RECV Pipeline accept success");
      }
    }

    MopErrCode ret = mopPipelineRecv_->recvData(readPack, &readPack.length);

    if (ret == MOP_PASSED) {
      std::vector<uint8_t> data(readPack.data, readPack.data + readPack.length);
      {
        std::lock_guard<std::mutex> lockGuard_(recvQueueMutex_);
        recvQueue_.push(data);
      }

    } else if (ret == MOP_CONNECTIONCLOSE) {
      MopErrCode closeErrCode = vehicle_->mopServer->close(mopPipelineRecv_->getId());

      LoggerManager::GetLogger("driver")->info("[FcIODjiImpl] recv pipeline disconnected");

      if (closeErrCode == MOP_PASSED) {
        mopPipelineRecv_ = nullptr;
      } else {
        continue;
      }

    } else if (ret == MOP_TIMEOUT) {
      LoggerManager::GetLogger("driver")->info("[FcIODjiImpl] readTimeout occured");
    } else {
      LoggerManager::GetLogger("driver")->info("[FcIODjiImpl] recv data from dji pipeline fail. ret code : {}", ret);

      std::shared_ptr<ReasonDataObject> dto = std::make_shared<ReasonDataObject>();
      dto->SetStatusCode(Define::NIV_MODE_SAFE_HOLD);  // TODO gen comm MOP_RECV_FAIL
      ServiceManager::Call("GoToSafeHoldService", dto);
    }
  }
}
```

dji-dev, 2021-12-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Yes, I found the "recv" code, but the logs shows that you sent a disconnect. I want to confirm that this sending is directly related to the start of PSDK 3.0 application? It seems to be recorded by your code.
 [image]
Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team​
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-14:
I didn't send disconnect
In recv data api already returned CONN CLOSE.

```
  } else if (ret == MOP_CONNECTIONCLOSE) {
      MopErrCode closeErrCode = vehicle_->mopServer->close(mopPipelineRecv_->getId());

      LoggerManager::GetLogger("driver")->info("[FcIODjiImpl] recv pipeline disconnected");

      if (closeErrCode == MOP_PASSED) {
        mopPipelineRecv_ = nullptr;
      } else {
        continue;
      }
```

> I want to confirm that this sending is directly related to the start of PSDK 3.0 application? It seems to be recorded by your code.

Yes. when PSDK connection negotiation finished and start streaming. 
The connection between MSDK and OSDK, which was waiting for the connection to be established. 
That connection accpeted & immediately disconnected.

```
[2021-12-14 16:08:40.295] [driver] [info] [FcIODjiImpl] MOP send Pipeline accept success
[2021-12-14 16:08:40.296] [driver] [info] [FcIODjiImpl] send data length : 423
[2021-12-14 16:08:40.299] [driver] [info] [FcIODjiImpl] MOP RECV Pipeline accept success
[2021-12-14 16:08:40.353] [driver] [info] [FcIODjiImpl] send pipeline disconnected
```

dji-dev, 2021-12-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Isn't this line from your code? I did not find this line of code, and it seems to be your record. Is it convenient for you to reproduce this problem based on PSDK 3.0 sample, OSDK4.1 sample? I still can't reproduce this issue.
 [image]​<br><a href='https://djisdksupport.zendesk.com/attachments/token/cYKJAwWD2OhrmXZ7t0Xvl7DrP' target='zd_attach'>inline1524116720.png</a>
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2021-12-15:
That line is from my code. when recv or send api returned 'MOP_CONNECTIONCLOSE'

OpusK, 2022-01-03:
HI, @dji-dev.  Happy new year ;)

> Isn't this line from your code? I did not find this line of code, and it seems to be your record. Is it convenient for you to reproduce this problem based on PSDK 3.0 sample, OSDK4.1 sample? I still can't reproduce this issue.

Could you please let me know examples you recommend?
- PSDK3.0 : dji_sdk_demo_linux
- OSDK4.1 : ??
  - You've been explained above that you have modified a existing sample.
- MSDK4.15.1 : ??

Thank you.

dji-dev, 2022-01-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

- PSDK3.0 : dji_sdk_demo_linux

- OSDK 4.1 please use the original sample
       https://github.com/dji-sdk/Onboard-SDK/blob/master/sample/platform/linux/mop/om_download_sample.cpp

- MSDK sample:(both 4.15.1 and 4.16 can be used for testing, MOP related functions are not updated)
      https://github.com/DJI-Mobile-SDK-Tutorials/Android-MOPDemo

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-05:
Hi, @dji-dev 

Did you have some replaceable samples for iOS? (I could not find it in GitHub repo)

dji-dev, 2022-01-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>https://github.com/dji-sdk/Mobile-SDK-iOS/tree/master/Sample Code/ObjcSampleCode/DJISdkDemo/Demo/Pipelines
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-06:
Hi, @dji-dev 

> 3、wait the PSDK 3.0 application "waiting payload negotiate finished" and then connects MSDK MOP to the OSDK 4.1, it works fine.

If we keep the above order, there is no problem, but when we start the PSDK while running the MSDK example APP, the following error (we have experienced) will come out.
```
[0.000][core]-[Info]-[DjiCore_Init:89) Payload SDK Version : V3.0.0-beta.0-build.1440
[0.001][utils]-[Debug]-[DjiCore_RootTask:131) work execution time:
[0.001][utils]-[Debug]-[DjiCore_RootTask:132) name	                execution time (ms)	  max execution time (ms)
[0.001][utils]-[Debug]-[DjiCore_RootTask:134) recorderWork                    	0       	0       
[0.001][utils]-[Debug]-[DjiCore_RootTask:134) flowControllerTask              	0       	0       
[0.011][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.011][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.032][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.052][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.072][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.092][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.118][linker]-[Debug]-[DjiLinker_DeInit:164) deinit linker success, protocol:2, channelId:0
[0.137][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.137][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.157][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.178][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.198][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.219][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.241][linker]-[Debug]-[DjiLinker_DeInit:164) deinit linker success, protocol:2, channelId:0
[0.263][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.263][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.270][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x03FD
[0.277][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C36
[0.277][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C43
[0.277][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C37
[0.283][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0xE5->0x65 0x03FD
[0.284][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x03FD len:1
[0.291][linker]-[Debug]-[DjiLinker_DeInit:164) deinit linker success, protocol:2, channelId:0
[0.291][adapter]-[Info]-[DjiAccessAdapter_Init:149) Identify aircraft series is Matrice 300 Series
[0.291][adapter]-[Info]-[DjiAccessAdapter_Init:166) Identify mount position type is Payload Port Type
[0.291][linker]-[Debug]-[DjiCommand_Init:155) config list channel id:2
[0.309][linker]-[Debug]-[DjiLinker_Init:124) init linker success, protocol:2, channelId:0
[0.309][linker]-[Debug]-[DjiCommand_Init:164) add linker node count:1
[0.309][linker]-[Debug]-[DjiCommand_RegRecvCmdHandler:599) Reg command handle list success, index:0
[0.309][adapter]-[Info]-[DjiPayloadNegotiate_Start:199) Waiting payload negotiate finished.
[0.772][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0x04 0x3C10
[0.772][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x3C10 len:64
[0.772][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x3C10 len:64
[0.776][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C36
[0.776][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C43
[0.776][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x65->0xE5 0x3C37
[0.777][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x3C36 len:0
[0.777][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x3C36 len:0
[0.777][linker]-[Debug]-[DjiCommand_DealCmd:821) received unsupport request cmd that need ack, cmdset: 0x3C cmdid: 0x36
[0.779][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x3C43 len:0
[0.779][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x3C43 len:0
[0.779][linker]-[Debug]-[DjiCommand_DealCmd:821) received unsupport request cmd that need ack, cmdset: 0x3C cmdid: 0x43
[0.780][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x3C37 len:16
[0.780][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x3C37 len:16
[0.780][linker]-[Debug]-[DjiCommand_DealCmd:821) received unsupport request cmd that need ack, cmdset: 0x3C cmdid: 0x37
[0.944][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x34->0x13 0x0087
[0.945][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x0087 len:4
[0.946][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x0087 len:4
[0.946][adapter]-[Debug]-[DjiPayloadNegotiate_VersionHandle:363) Payload negotiate use dynamic ip config
[0.946][linker]-[Debug]-[DjiProtocol_v1Pack:115) V1-Send: 0x13->0x34 0x0087
[0.978][linker]-[Debug]-[DjiProtocol_v1Unpack:192) V1-Recv: 0x34->0x13 0x0087
[0.978][linker]-[Debug]-[DjiLinker_RecvMsg:217) recv msg header success, cmd:0x0087 len:96
[0.978][linker]-[Debug]-[DjiCommand_RecvWork:958) recv msg from linker, id:0x0087 len:96
[0.978][adapter]-[Debug]-[DjiDeviceInfoParse_ParseCommand:389) command:3
[... cut ...]
[27.308][linker]-[Error]-[DjiLinker_RecvTask:261) send msg to queue error
```
In fact, we(or our customer) are starting the PSDK and OSDK applications, while we run MSDK App first to use the product first.
In other words, 
1. Run MSDK APP first
2. Boot Drone (Then, OSDK APP and PSDK APP began at the same time)

Do we have to keep the order you mentioned? (PSDK negotiate finished -> OSDK & MSDK) 
Do you have a document that describes the relationships and sequences of Pipeline? (I have not found it yet)

dji-dev, 2022-01-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the inconvenience. We have reproduced this problem and have submitted it to the relevant department for further confirmation and to fix. We  suggest that you follow the above steps to use temporarily.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-06:
understood.

Please let me know when Fix.
Thank you!

dji-dev, 2022-01-19:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The issue IOS MSDK sample and PSDK 3.0 connections, has been fixed, please update the PSDK 3.0 from the github.
https://github.com/dji-sdk/Payload-SDK

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

OpusK, 2022-01-25:
This is still working well.
Thanks for your support :)

OpusK, 2022-02-07:
Hi, @dji-dev  
Could you tell me the next release plan that contains this patch?

dji-dev, 2022-02-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57595'>Zendesk ticket #57595</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue has been fixed on the PSDK master branch. In order to improve the bug fix and push, if there is a bug in the PSDK itself, it will be directly pushed to the github master branch, and the version will no longer be released separately.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

closed 2022-10-20 by choiwooseok