dji-sdk/Onboard-SDK #856: undefined reference to `FlightSample::FlightSample(DJI::OSDK::Vehicle*)'

highfreq, 2021-12-09:
Please help me in linking what is missing when compilint only flight-control.
Cmake works but when i compile i get the error : undefined reference to `FlightSample::FlightSample(DJI::OSDK::Vehicle*)'

Could you please let me know what to add to CMakeLists.txt to have it working?

Please DO NOT link me to a useless github or forum answer, i already went troygh all of them and didn't find an answer.

dji-dev, 2021-12-09:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/57604'>Zendesk ticket #57604</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

highfreq, 2021-12-09:
I posted this question on every dji tech forum and i can't get an answer whatsoever. I think it is just some link missing in CMakeLists.txt but it looks like a secret and i can't get any help.

dji-dev, 2021-12-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/57604'>Zendesk ticket #57604</a>:<br><br>The developer's application development is beyond the scope of DJI SDK technical support, the link to the compiled file has been provided for this question, it is about the **basic compilation configuration**. It is recommended that you learn the basics of Linux C/C++ development first, and then port the OSDK for your application.

Just take the flight control sample as an example (copy telemetry CMakeLists.txt from github: https://github.com/dji-sdk/Onboard-SDK/issues/770)

update the CMakeLists.txt
 [image]
Compile:
​ [image]​​

update the source code:camera_manager_sync_sample.cpp
 [image]
​ [image]
<br><p><em><small>°°°</small></em></p>

highfreq, 2021-12-09:
Without making it longer than it needs to be....................just needed to add the following:

${ONBOARDSDK_SOURCE}/../sample/core/src/flight_sample.cpp

closed 2021-12-10 by dji-kyle

makk521, 2024-06-24:
where can i add this "${ONBOARDSDK_SOURCE}/../sample/core/src/flight_sample.cpp" ? In include_directories or other?

makk521, 2024-06-24:
> where can i add this "${ONBOARDSDK_SOURCE}/../sample/core/src/flight_sample.cpp" ? In include_directories or other?