dji-sdk/Onboard-SDK #857: Would be much better to have instruction to build each sample

highfreq, 2021-12-10:
Inside OSDK documentation would be much better for developers to have instructions to build each sample separately instead of having hundreds of people struggling to create the environment for each one. One, like me, coming from 30 years C# programming under windows would benefit a great deal in understanding how to create environment instead of all sample built together.
I think a lot of people have the same problem.
I also think documentation from dji is really poor seeing the dimensions of the company. Companies that big have documentations on their APK way better than they do. They really show very little respect to their customers.

dji-dev, 2021-12-10:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/57654'>Zendesk ticket #57654</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

highfreq, 2021-12-10:
This auto answer is really annoying, we know it is not your main consultation channel, stop saying it.