dji-sdk/Onboard-SDK #859: Using ActionExecEvent on M300

btsatoukzn, 2021-12-23:
hi,

We want to handle the execution of an action in an event callback.
However, the parameter was commented out.

dji_mission_type.hpp L:844-849
>   /*ID:0x30*/
>   // struct ActionExecEvent{
>   //   uint16_t actionId;
>   //   uint8_t preActuatorState;
>   //   uint8_t curActuatorState;
>   //   uint32_t result;
 >  // }ActionExecEvent;

So, when we uncommented these and ran the sample program 'djiosdk-waypoint-v2-sample' after building, we were able to refer to the parameters.

Why is the parameter commented out?
Is it okay to use this parameter?

dji-dev, 2021-12-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/58242'>Zendesk ticket #58242</a>:<br><br>This parameter can be uncommented and used.
<br><p><em><small>°°°</small></em></p>

btsatoukzn, 2021-12-24:
thank you for your prompt reply.

closed 2021-12-24 by btsatoukzn