dji-sdk/Onboard-SDK #861: About lifecycle of MopServer

OpusK, 2022-01-07:
Hi, 

This issue is related to #660.
The current situation is as follows.
1. Connect to the pipeline of MopServer and disconnect again.
2. Then, do not release the memory allocation of MopServer for re-connection
3. But, when reconnection is attempted, accept failed occurs. ( return code of accept function is 10 )

Unlike my test method, your origin example releases the MopServer's memory allocated after every disconnection and allocates memory again.
Does this are your intention?
Personally, I think it seems inefficient that the memory release & re-allocation must be repeated for every re-connection.

The code I use is as follows. (recommended in #855)
- MSDK : [iOS_pipelines_example](https://github.com/dji-sdk/Mobile-SDK-iOS/tree/master/Sample%20Code/ObjcSampleCode/DJISdkDemo/Demo/Pipelines)
- OSDK 4.1 : om_download_sample (some modified like below (please check MODIFIED) )
```cpp
static void* MopServerTask(void *arg)
{
  Vehicle *vehicle = (Vehicle *)arg;
  PipelineType type = UNRELIABLE;
  PipelineID id = TEST_OM_UNRELIABLE_PIPELINE_ID;

  /*! create mop server */
  MopServer *server = new MopServer();   // MODIFIED (just moved)

  for (;;) {
    // /*! create mop server */
    // MopServer *server = new MopServer();    // MODIFIED  (just commented)

    /*! connect pipeline */
    MopPipeline *OM_Pipeline = NULL;
    if (isReliable) {
      type = RELIABLE;
      id = TEST_OM_RELIABLE_PIPELINE_ID;
    }

    sampleFinish = false;
    MopErrCode acceptRet = MOP_PASSED;
    if (((acceptRet = server->accept(id, type, OM_Pipeline))
        != MOP_PASSED) || (OM_Pipeline == NULL)) {
      DERROR("MOP server accept failed, ret : %d", acceptRet);
      delete server;
      return NULL;
    } else {
      DSTATUS("accept successfully");
    }
    isAccept = true;

    /*! OSDK download file from  */
    if (isReliable) {
      OMDownloadFileTask(OM_Pipeline);
    } else {
      OMUnreliableTransTask(OM_Pipeline);
    }

    /*! Disconnect pipeline */
    if (server->close(id) != MOP_PASSED) {
      DERROR("MOP Pipeline disconnect pipeline(%d) failed", id);
    } else {
      DSTATUS("Disconnect mop pipeline id(%d) successfully", id);
    }
    sampleFinish = true;
    // delete server;      // MODIFIED  (just commented)
  }

  return NULL;
}
```

Thank you!

dji-dev, 2022-01-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/58899'>Zendesk ticket #58899</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This is a problem of code implementation, because there is ID management (dji_mop_server.cpp, MopServer::accept， pipelineMap) in server->accept, this part of code can be updated synchronously.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

choiwooseok, 2022-01-07:
Then  current solution is delete MopServer when disconnected is right??

I want to check delete vehicle->mopServer and new allocate is OK during flight.

 because MopServer instance is in vehicle object

OpusK, 2022-01-07:
> This is a problem of code implementation, because there is ID management (dji_mop_server.cpp, MopServer::accept， pipelineMap) in server->accept, this part of code can be updated synchronously.

pipelineMap[id] = p; called when server->accept() called.
pipelineMap.erase(id); called when server->close() called.
right?
If so, the above code is written as intended. Or..?

OpusK, 2022-02-07:
Any ideas? about @choiwooseok's and mine.

dji-dev, 2022-02-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/58899'>Zendesk ticket #58899</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、You can refer to the sample to repeat the execution of memory release & re-allocation when you re-connect.
2、If you do not want to repeatedly allocate mopserver resources, you can refer to the above reply and modify the implementation of the code in dji_mop_server.cpp as needed.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>