dji-sdk/Onboard-SDK #862: fatal error: dji_linux_environment.hpp: No such file or directory [DJI OSDK ROS 4.1.0]

fatbringer, 2022-01-11:
Hello, while trying to catkin_make for DJI OSDK ROS , i encountered this problem of :
```
fatal error: dji_linux_environment.hpp: No such file or directory
````

I have already installed DJI OSDK 4.1.0 successfully. 
I have also copied the dji_linux_environment.hpp and dji_linux_environment.cpp file to where the samples are located, but to no avail.

dji-dev, 2022-01-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59075'>Zendesk ticket #59075</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Which OSDK Linux version are you using？Please install the OSDK(linux) master branch and then compile the OSDK-ROS master branch.
If the error still exist, please provide the complete temminal logs.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

fatbringer, 2022-01-14:
I am using both OSDK 4.1.0 and OSDK ROS 4.1.0
the normal compilation is ok.

I want to create my own ROS executable. 
I have copied the dji_linux_environment.hpp and dji_linux_environment.cpp into the same folder as the samples. 

```
In file included from /media/dji/80GBstore/djisdk/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/samples/KN_target_tracker.cpp:33:0:
/media/dji/80GBstore/djisdk/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros/samples/dji_linux_helpers.hpp:36:37: fatal error: dji_linux_environment.hpp: No such file or directory
compilation terminated.
Onboard-SDK-ROS/src/dji_osdk_ros/samples/CMakeFiles/KN_target_tracker.dir/build.make:75: recipe for target 'Onboard-SDK-ROS/src/dji_osdk_ros/samples/CMakeFiles/KN_target_tracker.dir/KN_target_tracker.cpp.o' failed
make[2]: *** [Onboard-SDK-ROS/src/dji_osdk_ros/samples/CMakeFiles/KN_target_tracker.dir/KN_target_tracker.cpp.o] Error 1
CMakeFiles/Makefile2:5600: recipe for target 'Onboard-SDK-ROS/src/dji_osdk_ros/samples/CMakeFiles/KN_target_tracker.dir/all' failed
make[1]: *** [Onboard-SDK-ROS/src/dji_osdk_ros/samples/CMakeFiles/KN_target_tracker.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
````

Here are my includes
```
#include "dji_vehicle.hpp"
#include "std_msgs/String.h"
#include "dji_linux_environment.hpp"
#include "dji_linux_helpers.hpp"

#include <ros/ros.h>
#include <dji_osdk_ros/common_type.h>
#include <dji_osdk_ros/GimbalAction.h>
#include <dji_osdk_ros/CameraEV.h>
#include <dji_osdk_ros/CameraShutterSpeed.h>
#include <dji_osdk_ros/CameraAperture.h>
#include <dji_osdk_ros/CameraISO.h>
#include <dji_osdk_ros/CameraFocusPoint.h>
#include <dji_osdk_ros/CameraTapZoomPoint.h>
#include <dji_osdk_ros/CameraSetZoomPara.h>
#include <dji_osdk_ros/CameraZoomCtrl.h>
#include <dji_osdk_ros/CameraStartShootBurstPhoto.h>
#include <dji_osdk_ros/CameraStartShootAEBPhoto.h>
#include <dji_osdk_ros/CameraStartShootSinglePhoto.h>
#include <dji_osdk_ros/CameraStartShootIntervalPhoto.h>
#include <dji_osdk_ros/CameraStopShootPhoto.h>
#include <dji_osdk_ros/CameraRecordVideoAction.h>
````

MasterVols, 2023-02-01:
Was this ever resolved? I am having the same problem occur to me also.