dji-sdk/Onboard-SDK #863: Problem with flight control in the obtain control mode when the remote control is turned off (lost) on firmware 03.00.0101

IvanOrfanidi, 2022-01-12:
Dear Sir or Madam, 

During the operation we have experienced the following issue:
Problem with flight control in the obtain control mode when the remote control is turned off (lost) on firmware 03.00.0101.
When launching a mission while overriding control with the command obtain_release_control_authority
and the subsequent control of the drone by the command /dji_osdk_ros /flight_control_setpoint_generic. Control does not occur when the path is off or unavailable over the link.

The drone goes airborne. It flies to a sufficiently long distance at which the drone loses the remote control, but the mission execution continues , since it has been loaded into the drone earlier. When the mission is stopped and it is attempted to issue the / obtain_release_control_authority command with the parameter true, the drone will execute it without error. With further control via the /dji_osdk_ros /flight_control_setpoint_generic command with axes (sensor_msgs / Joy) parameters, the drone does not respond to them and remains in the air. Only after the second attempt to search for the RC, the drone responds only to its commands.

Please advise how to conduct a controlled flight with the command /dji_osdk_ros/flight_control_setpoint_generic if there is no connection with the remote control.

Regards,
Ivan

dji-dev, 2022-01-12:
Agent comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/59161'>Zendesk ticket #59161</a>:<br><br>尊敬的开发者，感谢您联系DJI 大疆创新
由于github不是我们主要的咨询渠道，您的问题可能跟进不及时。我们建议您通过填写表单（ https://djisdksupport.zendesk.com/hc/zh-cn/requests/new ）向我们反馈问题。或者您也可以在论坛发帖，与其它开发者交流。论坛链接：https://djisdksupport.zendesk.com/hc/zh-cn/community/topics

Dear developer, thank you for contacting DJI.
Since github is not our main consultation channel, your questions may not be followed up in time. We recommend that you fill in the form (https://djisdksupport.zendesk.com/hc/en-us/requests/new) to report problems to us. Or you can post in the forum to communicate with other developers. Forum link: https://djisdksupport.zendesk.com/hc/zh-cn/community/topics
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-01-13:
Thank you dear Friend!
We will check.

IvanOrfanidi, 2022-01-14:
It's okay! Thank you dear friend!
Tell me if there is a difference between giving the command "/obtain_release_control_authority" (in true) before and after turning off (lost) the remote controller?

closed 2022-01-14 by IvanOrfanidi

dji-dev, 2022-01-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59161'>Zendesk ticket #59161</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

"/obtain_release_control_authority" (in true) is used for OSDK to obtain control authority.
If OSDK does not obtain control, when the RC is disconnected, the drone will perform RC disconnection actions (hover/land/return setting via Pilot APP).

OSDK obtains control, and DJI Assistant2 enable SDK failsafe action, when RC is disconnected, OSDK will take over control authority.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-01-14:
Thank you very much!