dji-sdk/Onboard-SDK #864: Drone Arm Control

IvanOrfanidi, 2022-01-21:
Hello Dear friend.
Tell me, I'm trying to turn off the drone's engines to land it in an emergency.
The drone is in the air.
Executing the command: **/dji_osdk_ros/drone_arm_control 0**
But it's a mistake to come back: **ERROR: Service [/dji_osdk_ros/drone_arm_control] is not available.**
This is necessary to turn off the drone from the parachute data.

How do I turn off the engines when the drone is in flight?

IvanOrfanidi, 2022-01-21:
On firmware 03.00.0101.

dji-dev, 2022-01-24:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59580'>Zendesk ticket #59580</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Emergency stop during flight, please use API:
killSwitch

OSDK ROS is developed with the OSDK (Linux) API, and this API is defined in the OSDK Linux
https://github.com/dji-sdk/Onboard-SDK/blob/d08abe61768f50486ff10bca991c524e6308a011/osdk-core/api/inc/dji_flight_controller.hpp#L648

Note:
This API will automatically stop the propeller and crash. It is recommended to use it after full simulation and verification in the simulator to ensure safety.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-01-24:
Thank You Dear Friend. I used:
**_auto killswitchclient = n.ServiceClient<dji_osdk_ros::killswitch>("/kill_switch");_**

closed 2022-01-24 by IvanOrfanidi