dji-sdk/Onboard-SDK #865: Get status GPIO Manifold 2G

IvanOrfanidi, 2022-01-24:
Hello Dear Friend!
Tell me how do I get the status of the port I/O?

[image]
Or
[image]

Can I use ports just to receive logic signals?

There are services: 
_/dji_sdk/mfio_config _ 
_/dji_sdk/mfio_set_value_

But I don't understand which channel is needed.

How do I get the port status?

I'm sorry if the question is rather stupid, but I haven't found any documentation on them.

Is it possible to find out the status of the port directly, not through ROS?

Thanks for understanding!

dji-dev, 2022-01-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/59651'>Zendesk ticket #59651</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This I/O PORT corresponds to the manifold2 interface, and the SDK does not provide the manifold2 I/O control.
The MFIO function provided by the OSDK is the the drone MFIO function, but the M210/M300 no longer supports the OSDK to develop MFIO.
 [image]​

There is also no driver for the I/O port of the manifold2 on the SDK team. It is recommended that you consult the product after-sales directly: support@dji.com.

Please note that how to use the I/O port of the Manifold2.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/mkelEmM7eEP8P331aTf0N8Na0' target='zd_attach'>inline-35033442.png</a>
<br><p><em><small>°°°</small></em></p>

IvanOrfanidi, 2022-01-25:
Oh, I get it! Thanks!

closed 2022-01-25 by IvanOrfanidi